tinygo-org/tinygo #2242: WebUSB Support

hbroer, 2021-11-07:
Hey,

are there any plans to support WebUSB? WebUSB is still draft but in my opinion the a very good way to let users configure a device. 

For the C++ universe there is TinyUSB by Adafruit which supports WebUSB.

* https://learn.adafruit.com/using-webusb-with-arduino-and-tinyusb
* https://github.com/hathach/tinyusb

The API is supportet by all major browsers but Firefox:

* https://developer.mozilla.org/en-US/docs/Web/API/USB

Background: I am a web frontend developer and not realy into microcontroller programming. I only know some basics and tried tinygo a year ago because C/C++ is absolutly not my world and gives me headache ;-) I also did some golang programming in the past thats why I have tinygo on my watchlist for a while now. I also want to get into it again for a project I began a while ago and never finished. This project needs a way to get configured and the only good way I see is WebUSB because that is the quickes and most efficient way I can imagine. The project sources, schematics and documentation can be published on gitlab and also a compiled version of the config tool (a SPA) which can be open directly from the gitlab page. No need for drivers or a native application to be downloaded. But my understanding of bare metal programming is not good enough to understand if this is possiblem with TinyGo out of the box, or if that needs an additional library etc. I also thought about that if it is possible, I might create a kind of framework/sdk for the UI, so everyone, even without typescript and html/css knowledge, could create a basic web UI for his device.

gedw99, 2021-12-26:
hey @hbroer 

Not sure what USB devices you want to access over WebUSB you want to access via web usb but there is a golang gamepad here:https://github.com/gioui/gio-x/pull/8

Its a new API for GIO and has not been merged yet.  It does have a driver for JS, but i have not tested yet if webUSB picks it up.

anyways it might help you....

aykevl, 2021-12-30:
Do you mean WebUSB from the browser or from the device? Those two are mostly separate things.

@ardnew has been working on improved USB support for microcontrollers, but it's not ready yet. I think WebUSB support could be added to it.

hbroer, 2022-01-02:
@aykevl I mean to access a microcontroller device by web browser over USB. The web browser side is not the problem.

I don't understand how Web USB hast to be implemented on the micro controller side. Maybe directly into TinyGo drivers or so? It would be really nice to have a easy high level API. The web browser implementation of the API should be easy.

I can go a little more in detail on what I think the API could look like this month and make a draft about it and post it here. 

@gedw99 I don't think that that commit has something to do with WebUSB. It looks like it is Gamepad support over USB? Not sure.

gedw99, 2022-11-14:
Hey @hbroer 

the gio link is for both Web and non web .  There are some more links in the repo and it looks like it’s going to move forward and be maintained , so hopefully will be useful for you .

it will probably be able to run with no dependencies on gio also , as others have been able to be .

hbroer, 2022-11-14:
@gedw99 I can't see any relation to WebUSB. 

I did not got any time to look deeper into this topic at all, but just to try to make it more clear:

Microcontroller runing a firmware build with TinyGo connected via USB to a standard PC. Then a WebApp written in Typescript hosted on a public webserver needs to connect to the device via WebUSB to configure it. I don't want to access standard devices via Go. I want a custom device with a Microcontroller with my custom firmware written in TinyGo be accessible via WebUSB by a Browser like Chrome without any native app running on that PC. That is something complete different from accessing a gamepad via Go.

Microcontroller (TinyGo firmware) ⇦WebUSB⇨ PC (Chrom Browser, Typescript WebApp)

https://en.wikipedia.org/wiki/WebUSB

To be able to do this it would help a lot if TinyGo would have a API to make it easy to listen for connections and messages send via WebUSB and also makes it easy to send messages via WebUSB to the WebApp (messages in this context means data packages which can be objects or data streams). The first step would be a raw WebUSB support by TinyGo. The second step would be to build a easy to use API on both sides (TinyGo and Typescript) to make data exchange as easy as possible. Then in a last step it would be easy to build a Framework to setup a custom tool for any custom programmed microcontroller to configure and access it via a Website. This makes most custom drivers build for this devices obsolete and also the old way to access a microcontroller via serial interface and native running apps to configure this devices in general. If done right this could even be a big selling point for TinyGo. But that's another story. The important part is WebUSB in general. The standard is over my head on the microcontroller side. I am a WebDeveloper and that's the main reason why I am in here because TinyGo is way easier to handle than all that C based environments.