tinygo-org/tinygo #2244: Calling a function from JavaScript

hajsf, 2021-11-10:
I'm trying to replicate the `wasm` file, but got the error:

[image]

The code I used in the JS file is;
```JavaScript
const go = new Go(); // Defined in wasm_exec.js

go.importObject.env = {
    'main.add': function(x, y) {
        return x + y
    }
    // ... other functions
}

const WASM_URL = 'main.wasm';

var wasm;

if ('instantiateStreaming' in WebAssembly) {
	WebAssembly.instantiateStreaming(fetch(WASM_URL), go.importObject).then(function (obj) {
		wasm = obj.instance;
		go.run(wasm);
	})
} else {
	fetch(WASM_URL).then(resp =>
		resp.arrayBuffer()
	).then(bytes =>
		WebAssembly.instantiate(bytes, go.importObject).then(function (obj) {
			wasm = obj.instance;
			go.run(wasm);
		})
	)
}

// Calling the multiply function:
console.log('multiplied two numbers:', wasm.exports.multiply(5, 3));
```

All others `go` files used as per your example [here](https://tinygo.org/docs/guides/webassembly/)

aykevl, 2021-11-10:
It looks like you are calling `wasm.exports.multiply` before the module has finished loading. Note that it is only loaded in the `then` function. For example, this might work (untested):

```js
WebAssembly.instantiateStreaming(fetch(WASM_URL), go.importObject).then(function (obj) {
	wasm = obj.instance;
	go.run(wasm);
	console.log('multiplied two numbers:', wasm.exports.multiply(5, 3));
})
```

hajsf, 2021-11-10:
> it is only loaded in the `then` function

Thanks, I rewrote it `async/await`:
```JavaScript
WASM_URL = "http://localhost:8080/www/main.wasm"

async function init(){
    const go = new Go();
    if ('instantiateStreaming' in WebAssembly) {
        const result = await WebAssembly.instantiateStreaming(
            fetch(WASM_URL), 
            go.importObject
        );
        go.run(result.instance);
    } else {
        const bytes = await fetch(WASM_URL);
        bytes.arrayBuffer()
        const result = await WebAssembly.instantiate(
            bytes,
            go.importObject
        );
        go.run(result.instance);
    }
}
```
And made sure it is loaded from the index by:
```JavaScript
<html lang="en">
<head>
    <title>WASM</title>
    <script src="http://localhost:8080/www/lib.js"></script>
    <!-- WASM -->
    <script src="http://localhost:8080/www/wasm_exec.js"></script>
    <script src="http://localhost:8080/www/loadWasm.js"></script>
</head>
<body>
</body>
<script>
    (async () => {
        await init();
        alert("Wasm had been loaded")
        console.log(multiply(5, 3));
    })().catch(e => {
        console.log(e);
    });
</script>
</html>
```

closed 2021-11-10 by hajsf