tinygo-org/tinygo #2251: How to support stm32f4 board with less peripherals (e.g. TIM)

jakub-m, 2021-11-11:
I want to add support for stm32f401re Nucleo board. The existing src/machine/machine_stm32f4.go for this family of chips uses definitions for peripherals that do not exist for stm32f401 family. For example, TIM13:

src/machine/machine_stm32f4.go:278:28: TIM13 not declared by package stm32

Looking at ./src/device/stm32/stm32f401.go I see that there TIM1 to TIM11.

Question: how to modify machine_stm32f4.go so it is still usable for different chips of stm32f4 family, even those with less peripherals? Should I maybe exclude stm32f401 from machine_stm32f4.go and re-implement everything, or should I make machine_stm32f4.go somehow more generic?

kenbell, 2021-11-11:
There's no elegant solution I'm aware of since Go doesn't allow conditional compilation inside a source file.

I think the best thing is to have `machine_stm32f4.go` only contain the strictly common stuff (so TIM1 through to TIM11), and copy these to both `machine_stm32f405.go` and `machine_stm32f407.go`:
* TIM13
* TIM14(?)
* registerUPInterrupt
* registerOCInterrupt

deadprogram, 2021-11-18:
I think the request was fulfilled in the new release, so closing. Please reopen if needed. Thanks!

jakub-m, 2021-11-18:
Can you direct me to what is actually implemented in the next release? I'd know if I need to implement something on my own or if everything is covered. Thanks!

aykevl, 2021-11-18:
I don't think this is "solved" in any way, rather: build tags are the way to go here. It's a bit painful but I think it's the only reasonable option.
In many cases, it's possible to define peripheral implementations in a central file and only list specific instances in chip-specific files. For example, the file machine_nrf528xx.go implements PWM (`type PWM`) for the chips in the nrf528xx family, and machine_nrf52840.go and machine_nrf52.go use those to define `PWM0` etc.

deadprogram, 2021-11-20:
Reopening since seems this is a question.

davecheney, 2026-10-24:
NUCLEO-F401RE support is now merged in #5597. Chip-specific peripheral definitions are separated using build tags, addressing the unavailable TIM13 reference reported here.

This example builds successfully with the new target:

```go
package main

import (
    "machine"
    "time"
)

func main() {
    machine.LED.Configure(machine.PinConfig{Mode: machine.PinOutput})
    for {
        machine.LED.High()
        time.Sleep(500 * time.Millisecond)
        machine.LED.Low()
        time.Sleep(500 * time.Millisecond)
    }
}
```

```sh
tinygo build -target=nucleo-f401re -o blinky.elf main.go
```

This was a compile check, not a hardware test. The PR author also reports successful on-board LED and peripheral tests. Please open a new issue if any missing features or problems remain.

closed 2026-10-24 by davecheney