tinygo-org/tinygo #2259: rp2040: I2C not ACKing upon succesful read transaction

soypat, 2021-11-13:
RP2040 Transaction Write to *Address 0x40*, @ 0x00, and Read  @ 0x21
[image]

Same transaction on Arduino Mega 2560:
[image]

Notice last ACK bit is set on the Mega, not so on RP2040

deadprogram, 2021-12-07:
Is anyone looking into this?

soypat, 2022-03-25:
@ysoldak This may interest you.
I think I may have found the root cause. See [`func (i2c) Tx()`](https://github.com/tinygo-org/tinygo/blob/a5d905f19ba5008a88d4594bfa816ed6de1df7d0/src/machine/machine_rp2040_i2c.go#L77), this line should be changed to 
```go
if err := i2c.tx(uint8(addr), w, len(r) > 0, timeout); nil != err {
```
So that `nostop=true`, thus stop condition is not issued if there is a transaction pending on the read buffer.

Making the change breaks [`disable()`](https://github.com/tinygo-org/tinygo/blob/a5d905f19ba5008a88d4594bfa816ed6de1df7d0/src/machine/machine_rp2040_i2c.go#L183), presumably mid-transaction during `rx` call. The best approach would seem to be to combine `tx` and `rx` functions into a single function with two loops which disables the peripheral a single time during the transaction. I'll see if I fix it these coming days.

ysoldak, 2022-03-25:
@soypat thanks for looking at it, happy to help testing

soypat, 2022-03-26:
Alright, here's the first revision. seems to be working and I can confirm the spurious STOP condition when a read is pending is gone.
<details>

```go
func (i2c *I2C) Tx(addr uint16, w, r []byte) error {
	return i2c.txrx(uint8(addr), w, r)
}

func (i2c *I2C) txrx(addr uint8, tx, rx []byte) (err error) {
        // timeout in microseconds.
	const timeout = 40 * 1000 // 40ms is a reasonable time for a real-time system.
	deadline := ticks() + timeout
	if addr >= 0x80 || isReservedI2CAddr(addr) {
		return ErrInvalidTgtAddr
	}
	txlen := len(tx)
	rxlen := len(rx)
	// Quick return if possible.
	if txlen == 0 && rxlen == 0 {
		return nil
	}

	err = i2c.disable()
	if err != nil {
		return err
	}
	i2c.Bus.IC_TAR.Set(uint32(addr))
	i2c.enable()
	abort := false
	var abortReason uint32

	for txCtr := 0; txCtr < txlen; txCtr++ {
		if abort {
			break
		}
		first := txCtr == 0
		last := txCtr == txlen-1 && rxlen == 0
		i2c.Bus.IC_DATA_CMD.Set(
			(boolToBit(first && i2c.restartOnNext) << rp.I2C0_IC_DATA_CMD_RESTART_Pos) |
				(boolToBit(last) << rp.I2C0_IC_DATA_CMD_STOP_Pos) |
				uint32(tx[txCtr]))
		for !i2c.interrupted(rp.I2C0_IC_RAW_INTR_STAT_TX_EMPTY) {
			if ticks() > deadline {
				return errI2CWriteTimeout // If there was a timeout, don't attempt to do anything else.
			}
		}

		abortReason = i2c.getAbortReason()
		if abortReason != 0 {
			i2c.clearAbortReason()
			abort = true
		}
		if abort || last {
			// If the transaction was aborted or if it completed
			// successfully wait until the STOP condition has occured.

			// TODO Could there be an abort while waiting for the STOP
			// condition here? If so, additional code would be needed here
			// to take care of the abort.
			for !i2c.interrupted(rp.I2C0_IC_RAW_INTR_STAT_STOP_DET) {
				if ticks() > deadline {
					return errI2CWriteTimeout
				}
			}
			i2c.Bus.IC_CLR_STOP_DET.Get()
		}
	}
	if rxlen > 0 && !abort {
		for rxCtr := 0; rxCtr < rxlen; rxCtr++ {
			first := rxCtr == 0
			last := rxCtr == rxlen-1
			for i2c.writeAvailable() == 0 {
			}
			i2c.Bus.IC_DATA_CMD.Set(
				boolToBit(first && i2c.restartOnNext)<<rp.I2C0_IC_DATA_CMD_RESTART_Pos |
					boolToBit(last)<<rp.I2C0_IC_DATA_CMD_STOP_Pos |
					rp.I2C0_IC_DATA_CMD_CMD) // -> 1 for read
			for !abort && i2c.readAvailable() == 0 {
				abortReason = i2c.getAbortReason()
				i2c.clearAbortReason()
				if abortReason != 0 {
					abort = true
				}
				if ticks() > deadline {
					return errI2CReadTimeout // If there was a timeout, don't attempt to do anything else.
				}
			}
			if abort {
				break
			}
			rx[rxCtr] = uint8(i2c.Bus.IC_DATA_CMD.Get())
		}
	}
	// From Pico SDK: A lot of things could have just happened due to the ingenious and
	// creative design of I2C. Try to figure things out.
	if abort {
		switch {
		case abortReason == 0 || abortReason&rp.I2C0_IC_TX_ABRT_SOURCE_ABRT_7B_ADDR_NOACK != 0:
			// No reported errors - seems to happen if there is nothing connected to the bus.
			// Address byte not acknowledged
			err = ErrI2CGeneric
		case abortReason&rp.I2C0_IC_TX_ABRT_SOURCE_ABRT_TXDATA_NOACK != 0:
			// Address acknowledged, some data not acknowledged
			fallthrough
		default:
			err = makeI2CAbortError(abortReason)
		}
	}
	return err
}
```

</details>

soypat, 2023-02-24:
It's odd, I've seen a extremely similar NACK when working with the raspberry pi 4B+ last week... Not sure if the NACK was the sensors fault or the ARM I2C peripheral... though I guess it would not surprise me if the child followed in the father's footsteps.

deadprogram, 2024-11-30:
Closing since #2743 appears to address this. Thanks!

closed 2024-11-30 by deadprogram