tinygo-org/tinygo #2261: Persistent storage

conotto, 2021-11-14:
Hi,
Are there any examples on persistent storage ?
Ideally using on onboard flash.

Thanks

soypat, 2021-11-14:
I don't think you could use the onboard flash with TinyGo yet. The Pico-SDK is able to do this with some heavy use of C compiler magic https://github.com/raspberrypi/pico-sdk/blob/master/src/rp2_common/hardware_flash/flash.c#L86. I'd suggest using EEPROM peripheral via SPI or I2C.

conotto, 2021-11-14:
You are correct, having used esp32 onboard eeprom api, i was under assumption that its possible to do the same, however after further research it seems that esp idf partitions the onboard storage hence allowing to use internal storage that way. I will use the i2c module instead for this. Thanks for your reply

deadprogram, 2021-11-18:
I think you are looking for https://github.com/tinygo-org/tinyfs

conotto, 2021-11-28:
I may use that, thanks alot

mytechnotalent, 2023-03-05:
@deadprogram I spent a few days working with the `Arduino Nano RP2040 Connect` and attempting to write a simple `Hello` string to flash without any luck.

In https://github.com/tinygo-org/tinyfs I tried flashing the example `simple-fatfs` without any success as it flashed but when trying to read from the usb to serial it provided no data

Is there a functioning example to write persistent data to the `nano-rp2040`?  Thanks in advance.

deadprogram, 2023-05-22:
`machine.Flash` will be released as part of 0.28 so tagging this to close on next release.

deadprogram, 2023-06-14:
This is part of the `v0.28` release so now closing this issue. Thanks!

closed 2023-06-14 by deadprogram

dowling-john, 2024-11-27:
Sorry to open this one again.

I am looking to store a configuration file along with the application and update this via serial, but there is no real documentation on how to store something like this using the flash package.

I would apreciate some pointers on how I might acheive this.

sago35, 2024-11-28:
The following might be helpful.

https://github.com/sago35/tinygo-keyboard/blob/v0.6.0/via.go#L302