tinygo-org/tinygo #2270: Unable to build ESP32 (index out of range)

sago35, 2021-11-15:
Unable to build esp32 after the following versions.
XTENSA=0 is specified in the CI, but I think this problem will recur if XTENSA=1 is not set.

```
$ tinygo build -o /tmp/out.hex --size short --target esp32-coreboard-v2 examples/blinky1
panic: runtime error: index out of range [65521] with length 18

goroutine 51 [running]:
github.com/tinygo-org/tinygo/builder.loadProgramSize({0xc001770fc0, 0xc001b89ce8}, 0xc002d61860)
        C:/tinygo/tinygo/builder/sizes.go:271 +0xa65
github.com/tinygo-org/tinygo/builder.Build.func6(0x0)
        C:/tinygo/tinygo/builder/build.go:650 +0x905
github.com/tinygo-org/tinygo/builder.jobWorker(0x0, 0x0)
        C:/tinygo/tinygo/builder/jobs.go:182 +0xad
created by github.com/tinygo-org/tinygo/builder.runJobs
        C:/tinygo/tinygo/builder/jobs.go:101 +0x1a5

$ tinygo version
tinygo version 0.21.0-dev-5792f3a1 windows/amd64 (using go version go1.17.3 and LLVM version 11.0.0)
```

aykevl, 2021-11-16:
> XTENSA=0 is specified in the CI

Not in all places: the build-linux job doesn't set `XTENSA=0` for example. Therefore, this does work in CI.

aykevl, 2021-11-16:
However, I can reproduce this issue locally.

aykevl, 2021-11-16:
Thanks for reporting this! I've made a PR to fix this: https://github.com/tinygo-org/tinygo/pull/2271

sago35, 2021-12-20:
This problem does not occur in 0.21.0.
It has been resolved.

dgryski, 2021-12-20:
Closing then.

closed 2021-12-20 by dgryski