tinygo-org/tinygo #2304: Using PWM tutorial fails to compile for Raspberry Pi Pico device

awonak, 2021-11-22:
I am following the tutorial using a Raspberry Pi Pico device. I am able to get the [Blinking LED](https://tinygo.org/docs/tutorials/blinky/) example to work, however the [Using PWM](https://tinygo.org/docs/tutorials/pwm/) tutorial fails to compile with the following error:

```bash
$ tinygo flash -target=pico
# pwm-blinky
main.go:12:22: NewPWM not declared by package machine
```

If I alter the sample code as shown below, I am able to successfully compile the script and get the PWM working as expected on the Pico.

```go
// pwm, err := machine.PWMPeripheral(pin)
// if err != nil {
// 	println(err.Error())
// }
pwm := machine.PWM4
```

It seems that the current implementation ([src](https://github.com/tinygo-org/tinygo/blob/dev/src/machine/machine_rp2040_pwm.go)) is missing the `NewPWM` function and instead has a similar looking function `PWMPeripheral`, but that returns the slice index of the PWM group, not a pointer to the initialized peripheral.

Perhaps a solutions could be to include the `NewPWM` function like this:

```go
// NewPWM calls PWMPeripheral to get the RP2040 PWM peripheral slice number that
// corresponds to the package's PWM0 throughout PWM7 handler.
func NewPWM(pin Pin) (pwm *pwmGroup, err error) {
	sliceNum, err := PWMPeripheral(pin)
	if err != nil {
		return nil, err
	}
	return getPWMGroup(sliceNum)
}
```

soypat, 2021-11-22:
It seems the tinygo site is outdated, as all `NewPWM` API has been removed as of the PWM interface refactor. I believe @aykevl was working on creating a better PWM page on tinygo.org to better reflect these changes. 

Technical details aside, I agree we are missing some sort of API to get the underlying PWM peripheral from a pin. Right now it is quite unwieldy to do this user-side. I'm unsure why this was removed in the first place and whether it would be OK to add it as you have shown or some other way.

deadprogram, 2023-08-18:
The tutorial was updated some time ago, so now closing. Thanks everyone!

closed 2023-08-18 by deadprogram