tinygo-org/tinygo #2310: nucleo-l432kc i2c appears to be nonfunctional

rayozzie, 2021-11-24:
I have been trying to use this nucleo board for I2c, to no avail.  

The nucleo-l432kc pins are wired as SCL PB6 (D5) and SDA PB7 (D4).

I have several devices on the i2c bus.

After repeatedly failing to be able to speak with the devices using machine.Tx(), I then loaded https://github.com/toyo/I2Cscan and tried scanning the bus for any devices.  None could be seen.

Without any rewiring, I wrote an Arduino sketch to do the same, and it worked perfectly.

I then removed the wires and hooked them instead to a feather-nrf52840, and used that target for tinygo.  It worked perfectly.

As such, my conclusion is that the nucleo-l432kc I2C implementation is not functioning properly.

Thank you for your consideration of the issue.

deadprogram, 2021-11-25:
@rayozzie can you try the `dev` branch of TinyGo with your board to see if it addressed your specific issue? Appears to work as expected on STM32F4 board on TinyHCI.

kenbell, 2021-11-25:
@deadprogram - there was a problem with i2c on the stm32f407 some time back?  I think we 'fixed' it by putting a delay in the app.  I wonder if something similar here - perhaps some part of the initialization code isn't waiting for something to complete?  In my test I was sending i2c traffic in a loop - don't know if perhaps the first transmission is missing?

rayozzie, 2021-11-25:
@deadprogram i'm trying to follow instructions to build dev branch so i can try it.  MacOS big sur.
Currently blocked by:

$ git clone --recursive https://github.com/tinygo-org/tinygo.git
Cloning into 'tinygo'...
...
Cloning into '/Users/rozzie/dev/tinygo/lib/musl'...
fatal: repository 'https://git.musl-libc.org/musl/' not found
fatal: clone of 'git://git.musl-libc.org/musl' into submodule path '/Users/rozzie/dev/tinygo/lib/musl' failed
Failed to clone 'lib/musl' a second time, aborting

$ cd tinygo

tinygo $ git checkout dev
M	lib/CMSIS
M	lib/avr
M	lib/binaryen
M	lib/cmsis-svd
M	lib/compiler-rt
M	lib/mingw-w64
M	lib/nrfx
M	lib/picolibc
M	lib/stm32-svd
M	lib/wasi-libc
Branch 'dev' set up to track remote branch 'dev' from 'origin'.
Switched to a new branch 'dev'

tinygo $ git submodule update --init
Cloning into '/Users/rozzie/dev/tinygo/lib/musl'...
fatal: repository 'https://git.musl-libc.org/musl/' not found
fatal: clone of 'git://git.musl-libc.org/musl' into submodule path '/Users/rozzie/dev/tinygo/lib/musl' failed
Failed to clone 'lib/musl'. Retry scheduled
Cloning into '/Users/rozzie/dev/tinygo/lib/musl'...
fatal: repository 'https://git.musl-libc.org/musl/' not found
fatal: clone of 'git://git.musl-libc.org/musl' into submodule path '/Users/rozzie/dev/tinygo/lib/musl' failed
Failed to clone 'lib/musl' a second time, aborting

tinygo $

aykevl, 2021-11-25:
Looks like musl is having trouble. Luckily you probably don't need musl, at least not as long as you're not building Linux binaries.
I recommend to clone tinygo without the `--recursive` flag and then running `git submodule update --init`.

rayozzie, 2021-11-25:
thx.  continuing to fight the good fight:

tinygo $ git submodule update --init
Submodule 'lib/CMSIS' (https://github.com/ARM-software/CMSIS.git) registered for path 'lib/CMSIS'
Submodule 'lib/avr' (https://github.com/avr-rust/avr-mcu.git) registered for path 'lib/avr'
Submodule 'lib/binaryen' (https://github.com/WebAssembly/binaryen.git) registered for path 'lib/binaryen'
Submodule 'lib/cmsis-svd' (https://github.com/tinygo-org/cmsis-svd) registered for path 'lib/cmsis-svd'
Submodule 'lib/compiler-rt' (https://github.com/llvm-mirror/compiler-rt.git) registered for path 'lib/compiler-rt'
Submodule 'lib/mingw-w64' (https://github.com/mingw-w64/mingw-w64.git) registered for path 'lib/mingw-w64'
Submodule 'lib/musl' (git://git.musl-libc.org/musl) registered for path 'lib/musl'
Submodule 'lib/nrfx' (https://github.com/NordicSemiconductor/nrfx.git) registered for path 'lib/nrfx'
Submodule 'lib/picolibc' (https://github.com/keith-packard/picolibc.git) registered for path 'lib/picolibc'
Submodule 'lib/stm32-svd' (https://github.com/tinygo-org/stm32-svd) registered for path 'lib/stm32-svd'
Submodule 'lib/wasi-libc' (https://github.com/CraneStation/wasi-libc) registered for path 'lib/wasi-libc'
Cloning into '/Users/rozzie/dev/tinygo/lib/CMSIS'...
Cloning into '/Users/rozzie/dev/tinygo/lib/avr'...
Cloning into '/Users/rozzie/dev/tinygo/lib/binaryen'...
Cloning into '/Users/rozzie/dev/tinygo/lib/cmsis-svd'...
Cloning into '/Users/rozzie/dev/tinygo/lib/compiler-rt'...
Cloning into '/Users/rozzie/dev/tinygo/lib/mingw-w64'...
Cloning into '/Users/rozzie/dev/tinygo/lib/musl'...
fatal: repository 'https://git.musl-libc.org/musl/' not found
fatal: clone of 'git://git.musl-libc.org/musl' into submodule path '/Users/rozzie/dev/tinygo/lib/musl' failed
Failed to clone 'lib/musl'. Retry scheduled
Cloning into '/Users/rozzie/dev/tinygo/lib/nrfx'...
Cloning into '/Users/rozzie/dev/tinygo/lib/picolibc'...
Cloning into '/Users/rozzie/dev/tinygo/lib/stm32-svd'...
Cloning into '/Users/rozzie/dev/tinygo/lib/wasi-libc'...
Cloning into '/Users/rozzie/dev/tinygo/lib/musl'...
fatal: repository 'https://git.musl-libc.org/musl/' not found
fatal: clone of 'git://git.musl-libc.org/musl' into submodule path '/Users/rozzie/dev/tinygo/lib/musl' failed
Failed to clone 'lib/musl' a second time, aborting

kenbell, 2021-11-25:
For me, this works: git clone git://git.musl-libc.org/musl/
this fails: git clone https://git.musl-libc.org/musl/

The submodule is configured as:
```
submodule.lib/musl.path=lib/musl
submodule.lib/musl.url=git://git.musl-libc.org/musl
```
(from `git config --file .gitmodules -l`)

Any chance you have a git config that's changing 'git://' into 'https://' (using the `insteadof` config option?)

rayozzie, 2021-11-25:
Can’t do more today because thanksgiving, but that is almost certainly the problem.  Thx for the debug assist.

rayozzie, 2021-11-25:
Ok, so I got the dev version build and cleanly installed [thanks].

$ tinygo
No command-line arguments supplied.
TinyGo is a Go compiler for small places.
version: 0.22.0-dev

However, now when I try to build the simple i2c scanner, I get the following:
../../../Library/Caches/tinygo/goroot-go1.16.5-1a5e73fa17769497a6324b7152c1ac20bacc16ab6b2aadb7ed99e128c5cddbe4-syscall/src/runtime/runtime_stm32_timers.go:16:2: package device/stm32 is not in GOROOT (/Users/rozzie/Library/Caches/tinygo/goroot-go1.16.5-1a5e73fa17769497a6324b7152c1ac20bacc16ab6b2aadb7ed99e128c5cddbe4-syscall/src/device/stm32)

(yes, i tried doing an rm -rf of the tinygo library cache)

deadprogram, 2021-11-25:
You probably need to run `make gen-device` to rebuild the device wrappers for the various processors.

rayozzie, 2021-11-25:
That was it. Thx. 

So, net-net, with kids climbing all over, I was able to build and run the l432kc i2c scanner under dev, and it still fails.

kenbell, 2021-11-25:
I've opened a bug (above) for scanning-type behaviour - it's definitely broken and probably affects a wide set of stm32 MCUs.  I see lots of dropped messages.

kenbell, 2021-11-25:
I found a timeout issue - the PR above should help make i2c scanning work.

PR #2312 (2021-11-24): stm32: pull-up on I2C lines
Fixes #2310

This changes the default for all STM32 (except stm32f103)
commit a6200920 (2021-11-24): stm32: pull-up on I2C lines

fixes #2310
closed 2022-01-26 by aykevl