tinygo-org/tinygo #2318: uart not working inside goroutine

charlie-haley, 2021-11-25:
I have a requirement to run UART code inside goroutines, when using the [echo example](https://github.com/tinygo-org/tinygo/blob/release/src/examples/echo/echo.go), modified to run in a goroutine, I get no reply from the serial port. Moving the code into the main function, it works perfectly

```
package main

import (
	"machine"
	"time"
)

type state struct {
	name      string
	waterTemp int
}

var (
	txPin = machine.UART0_TX_PIN
	rxPin = machine.UART0_RX_PIN
	uart  = machine.UART0
)

func main() {
	go myRoutine()
}

func myRoutine() {
	uart.Configure(machine.UARTConfig{TX: txPin, RX: rxPin})
	uart.SetBaudRate(9600)
	uart.Write([]byte("Hello World\n"))

	input := make([]byte, 64)
	i := 0
	for {
		if uart.Buffered() > 0 {
			data, _ := uart.ReadByte()

			switch data {
			case 13:
				// return key
				uart.Write([]byte("\r\n"))
				uart.Write([]byte("You typed: "))
				uart.Write(input[:i])
				uart.Write([]byte("\r\n"))
				i = 0
			default:
				// just echo the character
				uart.WriteByte(data)
				input[i] = data
				i++
			}
		}
		time.Sleep(10 * time.Millisecond)
	}

}
```

**Device: RP2040 (Pico)**

ebusto, 2021-11-26:
You'll need to block inside `main`, otherwise the program exits once `main` returns.

charlie-haley, 2021-11-26:
That was it 🤦‍♂️, thanks @ebusto!

closed 2021-11-26 by charlie-haley