tinygo-org/tinygo #2319: could not find wasm-opt, set the WASMOPT environment variable to override

ymzuiku, 2021-11-25:
In macOS Monterey 12.0.1, cpu m1 pro

main.go:

```go
package main

import "fmt"

func main() {
	fmt.Println("hello tinygo")
}
```

Use tinygo build:

```sh
$ tinygo build -o wasm.wasm -target wasm ./main.go
```

Results:

```
error: could not find wasm-opt, set the WASMOPT environment variable to override
```

kenbell, 2021-11-25:
If installing from source, you can do `git submodule update --init` then `make binaryen`.  I don't use WASM, but from why i understand asyncify was made the default scheduler, so needs wasm-opt.

ymzuiku, 2021-11-26:
> If installing from source, you can do `git submodule update --init` then `make binaryen`. I don't use WASM, but from why i understand asyncify was made the default scheduler, so needs wasm-opt.

I installing from brew:

```
brew tap tinygo-org/tools
brew install tinygo
```

kenbell, 2021-11-26:
Looks like the brew formula needs updating.  Does `brew install binaryen` fix it?

malmod, 2021-11-27:
> Looks like the brew formula needs updating. Does `brew install binaryen` fix it?

I had the same issue, and yes this fixed it! Thanks

deadprogram, 2021-11-27:
Does anyone have time for a PR to https://github.com/tinygo-org/homebrew-tools?

fgsch, 2021-11-29:
@deadprogram Sure, I will open a PR to add wasm-opt.

deadprogram, 2021-11-29:
It is getting built here https://github.com/tinygo-org/tinygo/blob/dev/.circleci/config.yml#L155-L160

Is the problem actually just one of the path not getting set? In other words, just adding something like the following to the TinyGo Homebrew recipe? (not tested)

```ruby
   libexec.install "bin/wasm-opt"
   (bin/"wasm-opt").write_env_script libexec/"wasm-opt",
      :WASMOPT => prefix
```

fgsch, 2021-11-29:
@deadprogram I believe https://github.com/tinygo-org/homebrew-tools/pull/16 is enough but I will test it shortly.

fgsch, 2021-11-29:
@deadprogram while I haven't tested the PR above building the formula, copying wasm-opt inside libexec/bin made the error go away, so I'm confident that PR should fix this issue. Copying wasm-opt to $prefix/bin is not needed and might overwrite an existing binaryen installation.

ymzuiku, 2021-11-29:
@kenbell 

> `brew install binaryen` fix it?

 Yes this fixed it! Thanks

closed 2021-11-29 by ymzuiku

deadprogram, 2021-11-29:
Confirmed that the updated brew formula that @fgsch works to correct this, and has been merged into the `master` branch in the Homebrew tools repo for TinyGo. Thank you!

LennyPenny, 2023-06-14:
I've had the same issue on fedora btw, had to manually install `binaryen`

iamshreeram, 2023-11-07:
@deadprogram, I just installed `tinygo` using homebrew. Here is the version information:
```
tinygo version 0.30.0 darwin/amd64 (using go version go1.21.3 and LLVM version 16.0.1)
```
However, I am still encountering the following error:
```
error: could not find wasm-opt, set the WASMOPT environment variable to override
```
Fortunately, the issue was resolved after installing `binaryen`. Can you confirm if the homebrew recipe has been updated to address this problem?