tinygo-org/tinygo #2323: Segfault in reflect.valueInterfaceUnsafe

dgryski, 2021-11-26:
This code is extracted ad minimized from the easyjson test suite:

```
package main

import (
	"fmt"
	"reflect"
)

type marshaler interface {
	MarshalJSON() ([]byte, error)
}

type testType interface {
	marshaler
}

type Arrays struct {
	ByteArray [3]byte
}

var arrayValue = Arrays{
	ByteArray: [3]byte{'a', 'b', 'c'},
}

func (v Arrays) MarshalJSON() ([]byte, error) {
	return nil, nil
}

func main() {
	var test = struct {
		Decoded testType
	}{
		&arrayValue,
	}

	v1 := reflect.New(reflect.TypeOf(test.Decoded).Elem()).Interface()
	fmt.Println(v1)
}
```

Both calling `fmt.Println` or `reflect.DeepEqual` as the test was doing causes a segfault in `reflect.valueInterfaceUnsafe`.  Printing out the typecode and value on entrance to this `valueInterfaceUnsafe` prints the following
```
0x00000000000027e5 0x000000010aa23060
0x000000000000013f 0x000000010aa23060
0x0000000000000049 0x000000010aa23060
0x0000000000000010 0x0000000000000060
```

This does not fail on `target=wasi`.

dgryski, 2021-12-14:
`encoding/binary` also fails with trying to set an array element.  I wonder if it's related also?

dgryski, 2021-12-16:
The problem here seems to be that reflect.New is creating an indirect value even though the result type (a [3]byte) fits into a  uintptr, and then things Get Confused during `Index` when the assumptions don't match up.

dgryski, 2021-12-16:
(This was the same encoding/binary issue.)

aykevl, 2021-12-17:
> The problem here seems to be that `reflect.New` is creating an indirect value even though the result type (a `[3]byte`) fits into a `uintptr`, [...]

`reflect.New` always returns a pointer, so this behavior seems correct to me.

dgryski, 2021-12-17:
Yes.  The issue was more `isIndirect` issues, also fixed by the last commit in #2395 .

commit 518745ea (2021-12-18): src/reflect: make sure indirect pointers are handled correctly

Fixes #2370
Fixes #2323
closed 2022-01-26 by aykevl