tinygo-org/tinygo #2329: gen-device-avr produce incomplete set of registers

zdima, 2021-11-29:
@aykevl, @deadprogram 

I was working with AVR timer and have noticed that some boards get incomplete set of register.
First, I notice this for artiny85, where the TIMSK register was missing.

It looks like the issue due to this code:
https://github.com/tinygo-org/tinygo/blob/06df31944c6af706d6cb8b30a2e92d19060436d2/tools/gen-device-avr/gen-device-avr.go#L216-L224

The artiny85.atdf declare TIMSK register twice. What happening here is, the first record removed from the 'first' peripheral, but never added at a new location.

Before I submit PR with the fix (by removing lines 217-223), can someone tell if there is another reason behind it, other than to avoid duplicates?

Thanks

zdima, 2021-11-29:
One more question. Why files generate by gen-device-avr don't have masks and all multi-bit fields are written as separate bits?

 If we look at ATtiny85.atdf, we can see following `USIWM` bitfield with mask 0x30:
```
    <module caption="Universal Serial Interface" name="USI">
      <register-group caption="Universal Serial Interface" name="USI">
...
        <register caption="USI Control Register" name="USICR" offset="0x2D" size="1">
          <bitfield caption="Start Condition Interrupt Enable" mask="0x80" name="USISIE"/>
          <bitfield caption="Counter Overflow Interrupt Enable" mask="0x40" name="USIOIE"/>
          <bitfield caption="USI Wire Mode Bits" mask="0x30" name="USIWM" values="COMM_USI_OP"/>
```

The generator code would write this field as a set of separate bits.
```
	USICR_USIWM0 = 0x10 // USI Wire Mode Bits
	USICR_USIWM1 = 0x20 // USI Wire Mode Bits
```

aykevl, 2021-11-29:
It looks like `TIMSK` is defined in two separate modules, with different bitfields. I think those two sets of bitfields need to be merged together. I don't know exactly why the code is as it is, it's probably buggy (both gen-device-avr and gen-device-svd are a bit of a mess).

> One more question. Why files generate by gen-device-avr don't have masks and all multi-bit fields are written as separate bits?

Honestly, I don't know. I've probably never implemented it because many registers have single bit bitfields.

closed 2021-12-24 by zdima