tinygo-org/tinygo #2339: issue with maps indexed by structs? golang/geo/s2 TestCellFaces fails

dgryski, 2021-12-01:
```
--- FAIL: TestCellFaces
    edge (0.000000000000000000000000, 0.000000000000000000000000, 0.000000000000000000000000) counts wrong, got 1, want 2
    edge (0.000000000000000000000000, 0.000000000000000000000000, 0.000000000000000000000000) counts wrong, got 1, want 2
    edge (0.000000000000000000000000, 0.000000000000000000000000, 0.000000000000000000000000) counts wrong, got 1, want 2
    edge (0.000000000000000000000000, 0.000000000000000000000000, 0.000000000000000000000000) counts wrong, got 1, want 2
```

These are entries from a map indexed by a struct.   Here is another program that demonstrates the map-of-structs problem, although I'm not 100% certain it's the same issue as the failing test.

```
package main

import "fmt"

type Point struct {
	X, Y, Z float64
}

func main() {
	m := make(map[Point]int)
	p := Point{1, 1, 1}
	m[p]++
	for k, v := range m {
		fmt.Printf("%v/%v/%v: %d\n", k.X, k.Y, k.Z, v)
	}
}
```

On tinygo, this shows that the map key does not have the expected value.
```
~/go/src/github.com/dgryski/wasmplay/cmd/structmap $ go run main.go
1/1/1: 1
~/go/src/github.com/dgryski/wasmplay/cmd/structmap $ ~/go/src/github.com/tinygo-org/tinygo/build/tinygo  run main.go
1.715e-320/2.3185533814e-314/0: 1
```

dgryski, 2021-12-01:
Update: seems like the value that's coming out is the `interface{}` type code (`0x1c`).  The float64 `reflect.Kind` is 14.  Type codes for basic types are that type shifted over one bit, which for float64 makes it 28 == 0x1c.  Using https://evanw.github.io/float-toy/ (or `math.Frombits`) we can see that for a float64, the bit pattern `0x1c` is indeed `+1.383384e-322`.

aykevl, 2021-12-07:
See https://github.com/tinygo-org/tinygo/pull/2353 which should fix this bug.

closed 2022-01-26 by deadprogram