tinygo-org/tinygo #2348: builder: picolibc rename can fail when run concurrently

niaow, 2021-12-05:
We do not currently have any form of inter-process cache locking. As a result, the directory rename when building picolibc can (rarely) fail when run concurrently. Example:
```
error: rename /home/niaow/.cache/tinygo/picolibc-thumbv6m-unknown-unknown-eabi-cortex-m0plus/include.tmp2615988351 /home/niaow/.cache/tinygo/picolibc-thumbv6m-unknown-unknown-eabi-cortex-m0plus/include: file exists
````

sago35, 2021-12-05:
At least as of #1355 , there were cases of crashes during build.
The reason for this is cache inconsistency between multiple tinygo processes.

https://github.com/tinygo-org/tinygo/pull/1355#issuecomment-826015110

aykevl, 2021-12-06:
I _think_ it would be safe to use the existing file if `os.Rename` doesn't work.

Slightly better would be to use a lock, and check again whether the target has already been built by another process. This avoids rebuilding libraries when there is a cached version already (and thus wasting CPU time).

niaow, 2021-12-15:
I split the locking piece into #2392. We still need both, but the locking piece is more complex as it requires restructuring a few parts of the build to avoid deadlocks.

closed 2022-01-26 by deadprogram