tinygo-org/tinygo #2354: regression: http-get example in tinygo-drivers fails to compile with dev branch

ysoldak, 2021-12-07:
wifinina http-get example in tinygo-drivers repository fails to compile with tinygo from dev branch head.
It compiles fine with latest tinygo release (0.21.0)

Offending change is somewhere between
3d73ee77d37683384277f1ea9cef5bb64d2a76f9
and
ec39a77d514a1959f38f33ff8de8e819664ad5b0

```
tinygo build -target=arduino-nano33 -size=full -o test.bin ./examples/wifinina/http-get                                                                                                            cd-test
# path/filepath
/usr/local/Cellar/go/1.17.2/libexec/src/path/filepath/path.go:382:16: Lstat not declared by package os
/usr/local/Cellar/go/1.17.2/libexec/src/path/filepath/match.go:249:18: Lstat not declared by package os
/usr/local/Cellar/go/1.17.2/libexec/src/path/filepath/match.go:324:16: Stat not declared by package os
/usr/local/Cellar/go/1.17.2/libexec/src/path/filepath/path.go:465:18: Lstat not declared by package os
/usr/local/Cellar/go/1.17.2/libexec/src/path/filepath/path.go:501:18: Lstat not declared by package os
/usr/local/Cellar/go/1.17.2/libexec/src/path/filepath/symlink.go:84:17: Lstat not declared by package os
```

ysoldak, 2021-12-07:
PR that fails CircleCi due to this:
https://github.com/tinygo-org/drivers/pull/346

dgryski, 2021-12-08:
Figuring the correct build tags for the `os` package for each platform is a nightmare :(

Why was this not caught when that (or another) PR was merged?

dankegel, 2021-12-08:
Looking...

dankegel, 2021-12-08:
Simpler reproducer:
```
package main

import (
	"tinygo.org/x/drivers/net/http"
)

func main() {
}
```

ysoldak, 2021-12-08:
I mean, no wonder filesystem stuff is not available on "baremetal", shall drivers code be fixed instead that been working by coincidence before? Or shall we have no-op Stat and Lstat stubs for baremetal?

dgryski, 2021-12-08:
I think `ErrNotImplemented` stubs for everything on baremetal is the way to go, until we create some sort of rudimentary in-memory file system or something...

deadprogram, 2021-12-08:
Deciding on filesystem for bare metal also depends on the existence of some device to provide this. See https://github.com/tinygo-org/tinyfs

aykevl, 2021-12-08:
I suspect ec9fd3fb38c5aa97345cae517d6a15204581b851 is to blame. Before that commit, `"path/filepath"` could import the os package as usual. After that commit, `os.Stat` was only available for `!baremetal,!js`.

Not sure how to best catch this. In this specific case, it would have helped to add `"path/filepath"` to testdata/stdlib.go - but that's not really a scalable solution.
One other option is to split the definition and the implementation, for example one generic file (without build tags) defines the exported function:

```go
// some useful comment here
func Stat(name string) (FileInfo, error) {
    return stat(name)
}
```

And then it is implemented another file like this:

```go
// +build <some build tags>

func stat(...) ... { ...}
```

In fact, ec9fd3fb38c5aa97345cae517d6a15204581b851 _almost_ uses this pattern but not quite because of the `!baremetal,!js` build tags.

I've used this pattern here as well: https://github.com/tinygo-org/tinygo/pull/2332

dkegel-fastly, 2021-12-08:
I'll give it a whirl tonight if nobody beats me to it.

aykevl, 2021-12-08:
> I think `ErrNotImplemented` stubs for everything on baremetal is the way to go, until we create some sort of rudimentary in-memory file system or something...

Essentially, we use a kind of filesystem abstraction for the os package so that packages can set their own implementation. This is useful for embedded devices that don't have an OS and instead use a library (package) to implement the filesystem. It's very similar to the new `io/fs.FS` interface.

I'm fine with returning an `ErrNotImplemented` error for stat functions on baremetal - as long as the functions exist to avoid compile errors.

deadprogram, 2021-12-10:
This issue has been addressed in `dev`. Now closing, please reopen if needed. Thank you everyone!

closed 2021-12-10 by deadprogram