tinygo-org/tinygo #2356: Zero and Negative Zero are not treated as equal in maps

dgryski, 2021-12-08:
```
~/go/src/github.com/dgryski/bug $ go run bug.go
-0.000000e+000 2
-0.000000e+000 2
true true
~/go/src/github.com/dgryski/bug $ ~/go/src/github.com/tinygo-org/tinygo/build/tinygo run bug.go
+0.000000e+000 1
-0.000000e+000 1
+0.000000e+000 1
-0.000000e+000 1
true true
~/go/src/github.com/dgryski/bug $ cat bug.go
package main

import "math"

type fstruct struct {
	x float64
}

func main() {
	var zero float64
	var negzero float64 = math.Copysign(zero, -1)

	m := make(map[fstruct]int)

	var f fstruct
	var negf fstruct
	negf.x = negzero

	m[f]++
	m[negf]++

	for k, v := range m {
		println(k.x, v)
	}

	m2 := make(map[float64]int)
	m2[zero]++
	m2[negzero]++

	for k, v := range m2 {
		println(k, v)
	}

	println(zero == negzero, f == negf)
}
```

dgryski, 2021-12-08:
I'm working on a PR for this.

aykevl, 2021-12-08:
Thanks!
Looks like the bug is in `hashmapInterfaceHash`.

dgryski, 2021-12-08:
Sadly my fix for this solve the one test case for `golang/geo/s2`, but that package fails later with a Bus Error.

commit 1903cf23 (2021-12-08): src/runtime: improve float/complex hashing

This allows positive and negative zero to hash to the same value,
as required by Go.

This is not perfect, but the best I could do without
revamping all the hash funtions to take a seed.

Fixes #2356
closed 2022-01-26 by aykevl