tinygo-org/tinygo #2360: panic using Dial from "tinygo.org/x/drivers/net"

RichardoC, 2021-12-08:
When attempting open a TCP socket via the following code, I'm getting a panic for a nil pointer dereference instead of getting a response from google (or a connection refused)

I'm running it on both https://play.tinygo.org/ and https://go.dev/play/

```golang
package main

import (
	"bufio"
	"fmt"
	"tinygo.org/x/drivers/net"
)

func main() {
	fmt.Println("Hello, TinyGo")
	conn, err := net.Dial("tcp", "google.com:80")
	fmt.Println("Managed to Dial ")
	if err != nil {
	    fmt.Println(err)
	}
	fmt.Fprintf(conn, "GET / HTTP/1.0\r\n\r\n")
	status, err := bufio.NewReader(conn).ReadString('\n')
	if err != nil {
	    fmt.Println(err)
	}
	fmt.Println(status)	

}
```

Output from https://play.tinygo.org/ 

```bash
Hello, TinyGo
panic: runtime error: nil pointer dereference
```

Output from https://go.dev/play/

```bash
Hello, TinyGo
panic: runtime error: invalid memory address or nil pointer dereference
[signal SIGSEGV: segmentation violation code=0x1 addr=0x38 pc=0x48e6e6]

goroutine 1 [running]:
tinygo.org/x/drivers/net.ResolveTCPAddr({0xe?, 0xc00003c638?}, {0x4a8a27?, 0xc0000749c0?})
	/tmp/gopath15359808/pkg/mod/tinygo.org/x/drivers@v0.18.0/net/net.go:242 +0x66
tinygo.org/x/drivers/net.Dial({0x4a7d26, 0x3}, {0x4a8a27?, 0x1?})
	/tmp/gopath15359808/pkg/mod/tinygo.org/x/drivers@v0.18.0/net/net.go:75 +0x48
main.main()
	/tmp/sandbox5213433/prog.go:13 +0x90

Program exited.
```

Expected output
```bash
Hello, TinyGo
Managed to Dial 
dial tcp: lookup google.com on 169.254.169.254:53: dial udp 169.254.169.254:53: connect: no route to host

Program exited.
```

sago35, 2021-12-08:
Perhaps you need to add adapter initialization.

wifinina
https://github.com/tinygo-org/drivers/blob/90a13a697bc51b41b878ffc9b0fb5ff201ef0731/examples/wifinina/tcpclient/main.go#L40-L57k

rtl8720dn
https://github.com/tinygo-org/drivers/blob/90a13a697bc51b41b878ffc9b0fb5ff201ef0731/examples/rtl8720dn/tcpclient/main.go#L44-L53

RichardoC, 2021-12-10:
Thanks, I'm not using an SBC for this but that's good to know. Instead I'm using the playgrounds, which initialisation is appropriate for them?

AlmogBaku, 2022-02-19:
how is it possible with WASM?

sago35, 2022-02-20:
So far, it does not work with WASM.
For this reason, it does not work in playgrounds.

If someone creates an implementation of net.DeviceDriver, it will work.
https://github.com/tinygo-org/drivers/blob/90a13a697bc51b41b878ffc9b0fb5ff201ef0731/net/driver.go#L3-L17

RichardoC, 2022-02-20:
Given WASM doesn't support anything other than [HTTP/websockets and WebRTC](https://gafferongames.com/post/why_cant_i_send_udp_packets_from_a_browser/) I assume this can't be implemented then?

AlmogBaku, 2022-02-24:
Just updating, that upon my research - this is also not supported in WASI right now.
See https://radu-matei.com/blog/towards-sockets-networking-wasi/

RichardoC, 2022-02-25:
Closing the issue as unsupported by upstream

closed 2022-02-25 by RichardoC