tinygo-org/tinygo #2361: RP2040 Temperature sensor support

spearson78, 2021-12-08:
The RP2040 chip includes a built in temperature sensor. This can be read using the lower level device/rp APIs but it would be helpful for newcomers to have access to the temperature sensor either directly in the machine package or the drivers package.

Here is the code I have been using in my project.

```
import (
	"device/rp"
)
func InitTempSensor() {

	rp.ADC.CS.SetBits(rp.ADC_CS_TS_EN)
}

func ReadTemp() float32 {
	rp.ADC.CS.SetBits(uint32(4) << rp.ADC_CS_AINSEL_Pos)
	rp.ADC.CS.SetBits(rp.ADC_CS_START_ONCE)

	for !rp.ADC.CS.HasBits(rp.ADC_CS_READY) {
	}

	// rp2040 uses 12-bit sampling, so scale to 16-bit
	raw := uint16(rp.ADC.RESULT.Get())

	conversion_factor := float32(3.3) / float32(1<<12)
	result := float32(raw) * conversion_factor
	temp := float32(27) - (result-float32(0.706))/float32(0.001721)

	return temp
}
```

deadprogram, 2021-12-10:
I would think that you should be able to use the exist ADC support in the RP2040 combined with the thermistor driver? I have not looked too closely, but see https://github.com/tinygo-org/drivers/blob/release/thermistor/thermistor.go

spearson78, 2021-12-10:
Thanks, I'll take a look.

spearson78, 2021-12-10:
The machine and thermistor packages assume that ADC is always performed through a machine.Pin. This is not the the case for the rp2040 temperature sensor it is an additional ADC channel. I could create a fake Pin for the RP2040 but any attempt to use this pin as a standard GPIO pin would either need to be protected against which would make all GPIO slower just for this special case or if we omit the checks would lead to undefined behavior.

I'm not sure that the assumption that ADC channels are always bound to a machine.Pin will hold true for all microcontrollers especially for things like internal temperature sensors.

I think even the Arduino has an internal temp sensor not connected to a pin, https://microchipdeveloper.com/8avr:avrtemp however it seems here that the Arduino concept of analog pins doesn't work and the ADC channel has to be read using lower level code. https://forum.arduino.cc/t/using-the-internal-temperature-sensor/8201

The options I see are to either

1. Cleanly separate ADC channels from Pins this would give tinygo access to all internal ADC channels for things like internal temp sensors and voltage references. I could then modify the thermistor driver to be channel rather than pin based.

2. implement machine specific functions (in the machine package) in tinygo for these internal ADC channels maybe with a set of standardized names to  make them somewhat portable across compatible machines

3. implement machine specific functions in the drivers package again maybe with a naming convention to make them somewhat portable.

I didn't mean to open a can of worms here I was just messing around with the rp2040 and needed some analog values to use for testing and thought the temp sensor would be a quick source of real world values to use.

I'm willing to make these changes for the boards I have access to (Pi Pico and Arduino Uno)

The good news is of course the snippet of code I provided works with no changes to either machine or drivers which is a good sign that the lower level APIs are OK.

closed 2022-04-29 by deadprogram