tinygo-org/tinygo #2367: tinygo test -target waasi compress/bzip2 fails because it can't find test file?

dkegel-fastly, 2021-12-10:
With today's dev:

```
$ tinygo test compress/bzip2
ok  	compress/bzip2	0.533s
$ tinygo test -target wasi compress/bzip2
panic: open testdata/e.txt.bz2: file not found
```
Here's the source line that fails:

https://github.com/golang/go/blob/8ff254e30ba9d9c13747fe42213c3f54b47c58e7/src/compress/bzip2/bzip2_test.go#L216

dkegel-fastly, 2021-12-10:
compress/flate fails somewhat similarly with -target wasi but not without.

dgryski, 2021-12-10:
The problem is that `wasmtime` runs the tests with `--dir=.`, but the the `compress` packages also need `--dir=..` to be able to read `../testdata/*`.

Building the tests by hand (using `-c`) and running `wasmtime` passing the appropriate flags allows it to pass.

```
~/go/src/go.googlesource.com/go/src/compress/bzip2 $ ~/go/src/github.com/tinygo-org/tinygo/build/tinygo test -c -target=wasi compress/bzip2
~/go/src/go.googlesource.com/go/src/compress/bzip2 $ wasmtime run ./bzip2.test --dir=. --dir=.. -- -test.v
=== RUN   TestReader
--- PASS: TestReader
=== RUN   TestBitReader
--- PASS: TestBitReader
=== RUN   TestMTF
--- PASS: TestMTF
    [1 0 2 3 4]
    [1 0 2 3 4]
    [0 1 2 3 4]
    [4 0 1 2 3]
    [0 4 1 2 3]
=== RUN   TestZeroRead
--- PASS: TestZeroRead
PASS
```

aykevl, 2021-12-11:
This probably needs the following patch to work:
```diff
diff --git a/main.go b/main.go
index 927f2a2b..9a6bb375 100644
--- a/main.go
+++ b/main.go
@@ -244,7 +244,6 @@ func runPackageTest(config *compileopts.Config, result builder.BuildResult, test
                        flags = append(flags, "-test.run="+testRunRegexp)
                }
                cmd = executeCommand(config.Options, result.Binary, flags...)
-               cmd.Dir = result.MainDir
        } else {
                // Run in an emulator.
                args := append(config.Target.Emulator[1:], result.Binary)
@@ -264,6 +263,7 @@ func runPackageTest(config *compileopts.Config, result builder.BuildResult, test
                }
                cmd = executeCommand(config.Options, config.Target.Emulator[0], args...)
        }
+       cmd.Dir = result.MainDir
        cmd.Stdout = os.Stdout
        cmd.Stderr = os.Stderr
        err := cmd.Run()
```

Didn't try it this time but I needed it previously to get tests to pass using `tinygo test -target=wasi`.

dgryski, 2021-12-11:
I think the problem is a bit more general than that. A number of the tests require access to multiple directories but we only pass a single directory via `--dir`. This may be a case of needing to have special commands for a number of the stdlib package tests. Being able to pass arguments to the emulator from the tinygo command line would also be useful.

aykevl, 2021-12-11:
In that case, I think it makes sense to give these tests not just access to the current package but to the entire Go module (where the standard library is a single Go module).

dgryski, 2021-12-11:
And also `/tmp`, ...

dgryski, 2021-12-14:
@aykevl Your proposed patch is now in my PR, and it does work for `compress/bzip2`.  We'll figure out a more general solution when we have a few more stdlib test failures because of it.

dgryski, 2021-12-18:
With `compress/flate`, we do indeed need to pass `--dir=. --dir=.. --dir=../..`.  It would certainly be possible to add enough `--dir` flags for each directory up to the module root.

dgryski, 2021-12-18:
Where is the module root directory stored?  Does it live pre-computed somewhere or does it need to be discovered by looking in directories starting from the package directory for `go.mod` ?

dgryski, 2021-12-23:
Renamed bug to cover the more general case.  `compress/bzip2` passes now since it only needs access to a single directory.  `compress/flate` needs access to multiple and that's what we need a solution for.

aykevl, 2021-12-28:
> Where is the module root directory stored? Does it live pre-computed somewhere or does it need to be discovered by looking in directories starting from the package directory for `go.mod` ?

It is part of the `go list -json` result, which is called in loader/goroot.go. The specific fields are `Root` and/or `Module` (not yet read from the returned JSON). So they don't need any new calls, just a way to read them from the already available data. Passing this through so that `tinygo test` can read them shouldn't be very complicated.
(Just ask if you need any help.)

aykevl, 2021-12-30:
You can probably implement it in a similar way as `MainDir` in `builder.BuildResult`.

closed 2022-01-26 by aykevl

codefromthecrypt, 2022-09-12:
I wonder if we support go:embed if we need to do this? ex use go:embed instead of mustLoadFile?

codefromthecrypt, 2022-09-12:
sorry I forgot these are in the go src not ours. nevermind.

PR #2387 (2021-12-20): tinygo: set cmd.Dir even when running emulators
This allows compress/bzip2 to pass with -target=wasi

Fixes #2367
commit 85031d6d (2021-12-20): tinygo: set cmd.Dir even when running emulators

This allows compress/bzip2 to pass with -target=wasi

Fixes #2367
PR #2455 (2022-01-15): pass more --dirs to wasmtime so it can read the entire module tree
This allows compress/flate to pass on -target=wasi out of the box.

Fixes #2367
commit c87bb0e9 (2022-01-15): pass more --dirs to wasmtime so it can read the entire module tree

This allows compress/flate to pass on -target=wasi out of the box

Fixes #2367