tinygo-org/tinygo #2368: tinygo test -target wasi compress/zlib fails early?

dkegel-fastly, 2021-12-10:
With today's dev on mac, compress/zlib seems to crash very early on target wasi:
```
$ tinygo test  compress/zlib
ok  	compress/zlib	0.633s
$ tinygo test -target wasi compress/zlib
Error: failed to run main module `/var/folders/v0/0k9hwftd29b18z9z7s2g2sfc0000gn/T/tinygo646659090/main`

Caused by:
    0: failed to invoke command default
    1: wasm trap: unreachable
       wasm backtrace:
           0: 0x78bbf - <unknown>!asyncify_stop_unwind
           1: 0x767a9 - <unknown>!tinygo_launch
           2: 0x6281 - <unknown>!(*internal/task.Task).Resume
           3: 0x1d248 - <unknown>!_start
       note: using the `WASMTIME_BACKTRACE_DETAILS=1` environment variable to may show more debugging information
       
FAIL	compress/zlib	0.043s
```

dgryski, 2021-12-10:
With optimizations off, this crashes in the conservative garbage collector.  With `-gc=leaking`, the tests pass.  Note that `wasmtime` also needs `--dir=..` to allow the test package to access `../testdata`.

```
~/go/src/go.googlesource.com/go/src/compress/zlib $ ~/go/src/github.com/tinygo-org/tinygo/build/tinygo test -target=wasi -c -opt=1 compress/zlib
~/go/src/go.googlesource.com/go/src/compress/zlib $ WASMTIME_BACKTRACE_DETAILS=1 wasmtime run ./zlib.test --dir=. --dir=.. -- -test.v
=== RUN   TestDecompressor
--- PASS: TestDecompressor
=== RUN   TestWriter
Error: failed to run main module `./zlib.test`

Caused by:
    0: failed to invoke command default
    1: wasm trap: out of bounds memory access
       wasm backtrace:
           0: 0x212e9 - runtime.markRoots
                           at /Users/dgryski/go/src/github.com/tinygo-org/tinygo/src/runtime/gc_conservative.go:414:11
           1: 0x20cd3 - runtime.markStack
                           at /Users/dgryski/go/src/github.com/tinygo-org/tinygo/src/runtime/gc_stack_portable.go:31:12
           2: 0x20b7e - runtime.GC
                           at /Users/dgryski/go/src/github.com/tinygo-org/tinygo/src/runtime/gc_conservative.go:355:11
           3: 0x1f47 - runtime.alloc
                           at /Users/dgryski/go/src/github.com/tinygo-org/tinygo/src/runtime/gc_conservative.go:288:7
           4: 0x5f742 - compress/flate.NewWriter
                           at /Users/dgryski/go/src/go.googlesource.com/go/src/compress/flate/deflate.go:670:6
           5: 0x5f9b1 - compress/flate.NewWriterDict
                           at /Users/dgryski/go/src/go.googlesource.com/go/src/compress/flate/deflate.go:685:22
           6: 0x9faa3 - (*compress/zlib.Writer).writeHeader
                           at /Users/dgryski/go/src/go.googlesource.com/go/src/compress/zlib/writer.go:132:42
           7: 0xa000d - (*compress/zlib.Writer).Write
                           at /Users/dgryski/go/src/go.googlesource.com/go/src/compress/zlib/writer.go:146:24
           8: 0xa3733 - compress/zlib.testLevelDict$1
                           at /Users/dgryski/go/src/go.googlesource.com/go/src/compress/zlib/writer_test.go:62:23
           9: 0xa33c6 - <goroutine wrapper>
                           at /Users/dgryski/go/src/go.googlesource.com/go/src/compress/zlib/writer_test.go:54:2
          10: 0xa5c81 - <unknown>!tinygo_rewind
          11: 0x9463 - (*internal/task.Task).Resume
                           at /Users/dgryski/go/src/github.com/tinygo-org/tinygo/src/internal/task/task_asyncify.go:112:17
          12: 0x29741 - runtime.scheduler
                           at /Users/dgryski/go/src/github.com/tinygo-org/tinygo/src/runtime/scheduler.go:171:11
          13: 0x295f6 - runtime.run
                           at /Users/dgryski/go/src/github.com/tinygo-org/tinygo/src/runtime/scheduler_any.go:24:11
          14: 0x29577 - _start
                           at /Users/dgryski/go/src/github.com/tinygo-org/tinygo/src/runtime/runtime_wasm_wasi.go:20:5

~/go/src/go.googlesource.com/go/src/compress/zlib $ ~/go/src/github.com/tinygo-org/tinygo/build/tinygo test -gc=leaking -target=wasi -c -opt=1 compress/zlib
~/go/src/go.googlesource.com/go/src/compress/zlib $ WASMTIME_BACKTRACE_DETAILS=1 wasmtime run ./zlib.test --dir=. --dir=.. -- -test.v
=== RUN   TestDecompressor
--- PASS: TestDecompressor
=== RUN   TestWriter
--- PASS: TestWriter
=== RUN   TestWriterBig
--- PASS: TestWriterBig
=== RUN   TestWriterDict
--- PASS: TestWriterDict
=== RUN   TestWriterReset
--- PASS: TestWriterReset
=== RUN   TestWriterDictIsUsed
--- PASS: TestWriterDictIsUsed
PASS
```

dgryski, 2021-12-13:
@niaow Any thoughts on this garbage collection issue?

niaow, 2021-12-14:
I just realized that we dont actually swap the stack chain on goroutine switch. Idk if that would cause this, but I will try implementing it.

niaow, 2021-12-14:
Hmm it seems that that was it.

niaow, 2021-12-14:
See #2388

dkegel-fastly, 2021-12-22:
still happening with today's dev...

niaow, 2021-12-22:
To clarify: the GC panic was fixed, the asyncify panic was not

niaow, 2021-12-30:
I found the source of the asyncify panic: the asyncify stack overflowed during unwind.

I am going to make a PR to bump the default stack size to 16KiB on wasm/wasi since this keeps happening.

dgryski, 2021-12-30:
Fixed by the stack-size bump.

closed 2022-01-26 by deadprogram