tinygo-org/tinygo #2369: tinygo test -target wasi os fails with unknown import env::chmod

dkegel-fastly, 2021-12-10:
With today's dev on mac:

```
$ tinygo test os
ok  	os	0.235s
$ tinygo test -target wasi os
Error: failed to run main module `/var/folders/v0/0k9hwftd29b18z9z7s2g2sfc0000gn/T/tinygo491621975/main`

Caused by:
    0: failed to instantiate "/var/folders/v0/0k9hwftd29b18z9z7s2g2sfc0000gn/T/tinygo491621975/main"
    1: unknown import: `env::chmod` has not been defined
FAIL	os	0.032s
```

aykevl, 2021-12-11:
Apparently wasi-libc doesn't define `chmod`.

dkegel-fastly, 2021-12-11:
I gather wasi-filesystem is the part of wasi that defines things like chmod, and hadn't gotten around to that function yet.

https://github.com/WebAssembly/WASI/blob/main/phases/ephemeral/docs.md mentioned but did not define a function path_permissions_set.

https://github.com/WebAssembly/wasi-filesystem/pull/36 appears to move a bunch of cheese, but interestingly mentions a function change_file_permissions_at.

dkegel-fastly, 2021-12-18:
I think this can be closed.  tinygo test -target wasi os gets past this.

closed 2021-12-18 by dkegel-fastly