tinygo-org/tinygo #2370: tinygo test -target wasi reflect fails with unimplemented: (reflect.Value).MapKeys()

dkegel-fastly, 2021-12-10:
Evidently reflect is not implemented as completely for wasi?

```
$ tinygo test reflect
ok  	reflect	0.109s
$ tinygo test -target wasi reflect
panic: unimplemented: (reflect.Value).MapKeys()
...
```

dgryski, 2021-12-10:
This is an odd one.  `reflect.Value.MapKeys()` *is* unimplemented, so I'm not sure why there's a difference here.  Looking at the code it seems that maybe wasi is not able to use one of the short circuits before falling back to actually comparing the map keys? https://github.com/tinygo-org/tinygo/blob/release/src/reflect/deepequal.go#L115

aykevl, 2021-12-10:
Reflect should be supported just as well on WASI as on anything else, so there must be a bug somewhere.

I've tested a few different targets (linux/arm, linux/386, cortex-m-qemu, riscv-qemu, wasm, wasi) and it appears to be a problem only for WebAssembly.

dkegel-fastly, 2021-12-11:
With https://github.com/tinygo-org/tinygo/pull/2350 applied to dev, on mac, joho/godotenv runs into it, too.  To reproduce:
```
$ git clone https://github.com/joho/godotenv
$ cd godotenv
$ tinygo test -c
$ lldb ./godotenv.test -- -test.v -test.run TestRoundtrip
(lldb) r
Process 19044 launched: '/Users/dkegel/src/tinygo-test-corpus/_corpus/joho/godotenv/godotenv.test' (x86_64)
=== RUN   TestRoundtrip
panic: unimplemented: (reflect.Value).MapKeys()
(lldb) bt
* thread #1, queue = 'com.apple.main-thread', stop reason = signal SIGABRT
    frame #2: 0x00007fff2029c406 libsystem_c.dylib`abort + 125
    frame #3: 0x0000000100008701 godotenv.test`runtime._panic + 46
    frame #4: 0x00000001000047bc godotenv.test`(reflect.Value).MapKeys + 16
    frame #5: 0x000000010000424f godotenv.test`reflect.deepValueEqual + 1336
    frame #6: 0x000000010001fe24 godotenv.test`github.com/joho/godotenv.TestRoundtrip + 663
```

dgryski, 2021-12-12:
I'm less concerned about the `godotenv` tests failing with the unimplemented `reflectValue.MapKeys()`, since as I said above that function really is unimplemented.  The confusing thing about the reflect tests thought is that `tinygo test reflect` should run the tinygo's reflect package tests, which pass, and the exact same code then fails under wasi.  That to me is a more concerning issue.

dgryski, 2021-12-12:
Huh.  Patching deepequal.go to print out the stages of the map comparision test gives the following:
```
dgryski@Damians-MBP-2 reflect % ../../build/tinygo test reflect
map
v1.IsNil() false v2.IsNil() false
v1.Len() 4437430384 v2.Len() 10
length failed
map
v1.IsNil() false v2.IsNil() false
v1.Len() 0 v2.Len() 0
v1.Pointer() 0x00000001153c9f00 v2.Pointer() 0x00000001153c9f00
pointers equal
ok  	reflect	0.192s
dgryski@Damians-MBP-2 reflect % ../../build/tinygo test -target=wasi reflect
map
v1.IsNil() false v2.IsNil() false
v1.Len() 0 v2.Len() 0
v1.Pointer() 0x00023e70 v2.Pointer() 0x00023f0c
checking map keys
panic: unimplemented: (reflect.Value).MapKeys()
Error: failed to run main module `/var/folders/_k/0tn37c6x0pz5jkcn933p25rc0000gn/T/tinygo2615195877/main`
```

Clearly something weird is going on.

dkegel-fastly, 2021-12-12:
as far as I can tell, godotenv isn't calling an unsupported function, it's just calling reflect.DeepEqual()...

dgryski, 2021-12-12:
No, but if godotenv is trying to compare maps that's going to fail. My concern (as indicated by the debugging output above) is why the first map appears to have 4437430384 entries when run under native but 0 under wasi. Makes me feel that the wasi one is actually correct here.

dgryski, 2021-12-12:
```
package main

import "reflect"

var cycleMap1, cycleMap3 map[string]interface{}

func init() {
/*
        cycleMap1 = map[string]interface{}{}
        cycleMap1["cycle"] = cycleMap1
        cycleMap3 = map[string]interface{}{}
        cycleMap3["different"] = cycleMap3
*/
}

func main() {
        cycleMap1 = map[string]interface{}{}
        cycleMap1["cycle"] = cycleMap1
        cycleMap3 = map[string]interface{}{}
        cycleMap3["different"] = cycleMap3

	var a interface{} = &cycleMap1
	var b interface{} = &cycleMap3
	println(reflect.ValueOf(a).Elem().Len())
	println(reflect.ValueOf(b).Elem().Len())
}
```

If the initialization of the maps happens in `main()`, the output is
```
0
0
```. 

If it happens in `init()`, the output is 
```
0
4308132024
```

Note that both of these are incorrect since the maps each have 1 element.

Under `target=wasi`, the output is
```
0
65920
```

So, still broken.

dgryski, 2021-12-12:
I wonder if this is related to slightly broken reflect stuff from https://github.com/tinygo-org/tinygo/issues/2323 ?

dgryski, 2021-12-13:
I think this is about as small I can I make it:
```
package main

import "reflect"

var cycleMap1 map[string]interface{}

func main() {
	cycleMap1 = map[string]interface{}{}
	cycleMap1["cycle"] = cycleMap1

	var a = &cycleMap1
	println(reflect.ValueOf(a).Elem().Len())
}
```

And note that this is still wrong because this prints `0` even though the map has size `1`.  So, something is going wrong with the `Elem()` lookup.  Removing the `&` from the assignment to `a`  and the `Elem()` call correctly prints 0.

@aykevl Any ideas?

aykevl, 2021-12-17:
Have you tried this with https://github.com/tinygo-org/tinygo/pull/2395? I think that PR will fix this bug. Or at least https://github.com/tinygo-org/tinygo/issues/2370#issuecomment-992713319.

dgryski, 2021-12-17:
Yes, #2395 was the PR I made to fix this issue.

commit 518745ea (2021-12-18): src/reflect: make sure indirect pointers are handled correctly

Fixes #2370
Fixes #2323
closed 2022-01-26 by aykevl