tinygo-org/tinygo #2374: Importing "image/color" breaks scheduler=tasks

cstrahan, 2021-12-10:
Importing `"image/color"` breaks `scheduler=tasks`:

```
$ tinygo version
tinygo version 0.21.0 linux/amd64 (using go version go1.17.1 and LLVM version 11.0.0)
```

```go
// main.go
package main

import _ "image/color"

func main() {
}
```

```
$ tinygo flash -target arduino-mega2560 -scheduler=tasks
Invalid bitcast
i8* bitcast (%runtime._interface (i16, i8*, i8*, i8*) addrspace(1)* @"image/color.rgbaModel" to i8*)
Invalid bitcast
i8* bitcast (%runtime._interface (i16, i8*, i8*, i8*) addrspace(1)* @"image/color.rgba64Model" to i8*)
Invalid bitcast
i8* bitcast (%runtime._interface (i16, i8*, i8*, i8*) addrspace(1)* @"image/color.nrgbaModel" to i8*)
Invalid bitcast
i8* bitcast (%runtime._interface (i16, i8*, i8*, i8*) addrspace(1)* @"image/color.nrgba64Model" to i8*)
Invalid bitcast
i8* bitcast (%runtime._interface (i16, i8*, i8*, i8*) addrspace(1)* @"image/color.alphaModel" to i8*)
Invalid bitcast
i8* bitcast (%runtime._interface (i16, i8*, i8*, i8*) addrspace(1)* @"image/color.alpha16Model" to i8*)
Invalid bitcast
i8* bitcast (%runtime._interface (i16, i8*, i8*, i8*) addrspace(1)* @"image/color.grayModel" to i8*)
Invalid bitcast
i8* bitcast (%runtime._interface (i16, i8*, i8*, i8*) addrspace(1)* @"image/color.gray16Model" to i8*)
Invalid bitcast
i8* bitcast (%runtime._interface (i16, i8*, i8*, i8*) addrspace(1)* @"image/color.yCbCrModel" to i8*)
Invalid bitcast
i8* bitcast (%runtime._interface (i16, i8*, i8*, i8*) addrspace(1)* @"image/color.nYCbCrAModel" to i8*)
Invalid bitcast
i8* bitcast (%runtime._interface (i16, i8*, i8*, i8*) addrspace(1)* @"image/color.cmykModel" to i8*)
```

cstrahan, 2021-12-10:
Minimal reproducer:

```go
package main

type wrappedFunc struct {
	f func()
}

var (
	interfaceContainingPointerToWrappedFunc interface{} = &wrappedFunc{dummyFunc}
)

func dummyFunc() {}

func main() {}
```

```
$ tinygo flash -target arduino-mega2560 -scheduler=tasks
Invalid bitcast
i8* bitcast (void (i8*, i8*) addrspace(1)* @main.dummyFunc to i8*)
fatal error: unexpected signal during runtime execution
[signal SIGSEGV: segmentation violation code=0x80 addr=0x0 pc=0x431b923]

runtime stack:
runtime.throw({0x463379d, 0x0})
        /usr/local/go/src/runtime/panic.go:1198 +0x71
runtime.sigpanic()
        /usr/local/go/src/runtime/signal_unix.go:719 +0x396

goroutine 58 [syscall]:
runtime.cgocall(0xdf3a80, 0xc0010f7b80)
        /usr/local/go/src/runtime/cgocall.go:156 +0x5c fp=0xc0010f7b58 sp=0xc0010f7b20 pc=0xaa37bc
tinygo.org/x/go-llvm._Cfunc_LLVMGetValueName(0x7f8414007478)
        _cgo_gotypes.go:6723 +0x4d fp=0xc0010f7b80 sp=0xc0010f7b58 pc=0xcbe80d
tinygo.org/x/go-llvm.Value.Name.func1({0xc000253598})
        /go/pkg/mod/tinygo.org/x/go-llvm@v0.0.0-20210325115028-e7b85195e81c/ir.go:706 +0x3f fp=0xc0010f7bb8 sp=0xc0010f7b80 pc=0xccd2bf
tinygo.org/x/go-llvm.Value.Name({0xc0002a0720})
        /go/pkg/mod/tinygo.org/x/go-llvm@v0.0.0-20210325115028-e7b85195e81c/ir.go:706 +0x19 fp=0xc0010f7bd0 sp=0xc0010f7bb8 pc=0xccd259
github.com/tinygo-org/tinygo/builder.Build.func1(0x0)
        /home/circleci/project/builder/build.go:327 +0xa78 fp=0xc0010f7f58 sp=0xc0010f7bd0 pc=0xdb71b8
github.com/tinygo-org/tinygo/builder.jobWorker(0x0, 0x0)
        /home/circleci/project/builder/jobs.go:191 +0xad fp=0xc0010f7fc0 sp=0xc0010f7f58 pc=0xdbf04d
github.com/tinygo-org/tinygo/builder.runJobs·dwrap·13()
        /home/circleci/project/builder/jobs.go:110 +0x2a fp=0xc0010f7fe0 sp=0xc0010f7fc0 pc=0xdbed0a
runtime.goexit()
        /usr/local/go/src/runtime/asm_amd64.s:1581 +0x1 fp=0xc0010f7fe8 sp=0xc0010f7fe0 pc=0xb03681
created by github.com/tinygo-org/tinygo/builder.runJobs
        /home/circleci/project/builder/jobs.go:110 +0x229

goroutine 1 [chan receive]:
github.com/tinygo-org/tinygo/builder.runJobs(0x45c30e0, 0xc0010820c0)
        /home/circleci/project/builder/jobs.go:142 +0x314
github.com/tinygo-org/tinygo/builder.Build({0x0, 0x0}, {0x461d2d2, 0x5}, 0xc000282db0, 0xc000c55680)
        /home/circleci/project/builder/build.go:743 +0x2d85
main.Flash({0x0, 0x0}, {0x0, 0x0}, 0x0)
        /home/circleci/project/main.go:317 +0x4e5
main.main()
        /home/circleci/project/main.go:1308 +0x1f3f

goroutine 10 [chan receive]:
github.com/tinygo-org/tinygo/builder.jobWorker(0x0, 0x0)
        /home/circleci/project/builder/jobs.go:188 +0x5e
created by github.com/tinygo-org/tinygo/builder.runJobs
        /home/circleci/project/builder/jobs.go:110 +0x229

goroutine 11 [chan receive]:
github.com/tinygo-org/tinygo/builder.jobWorker(0x0, 0x0)
        /home/circleci/project/builder/jobs.go:188 +0x5e
[... 74 lines cut ...]

goroutine 57 [chan receive]:
github.com/tinygo-org/tinygo/builder.jobWorker(0x0, 0x0)
        /home/circleci/project/builder/jobs.go:188 +0x5e
created by github.com/tinygo-org/tinygo/builder.runJobs
        /home/circleci/project/builder/jobs.go:110 +0x229
```

Note that this compiles just fine (the key difference is that I _don't_ take the address of the wrapped func):

```go
package main

type wrappedFunc struct {
	f func()
}

var (
	interfaceContainingWrappedFunc interface{} = wrappedFunc{dummyFunc}
)

func dummyFunc() {}

func main() {}
```

cstrahan, 2021-12-10:
This also compiles successfully:

```go
package main

type wrappedFunc struct {
	f func()
}

var (
	wrapped                                 wrappedFunc = wrappedFunc{dummyFunc}
	interfaceContainingPointerToWrappedFunc interface{} = &wrapped
)

func dummyFunc() {}

func main() {}
```

cstrahan, 2021-12-10:
Is there a way to override the standard libraries? If I could remove the offending functions/globals (I don't _think_ they're used), I could workaround this for now.

aykevl, 2021-12-10:
Thanks for the small reproducer! That really helps to find the precise bug location.

This should fix it: https://github.com/tinygo-org/tinygo/pull/2376

> Is there a way to override the standard libraries? If I could remove the offending functions/globals (I don't _think_ they're used), I could workaround this for now.

You can edit the files directly, use `go env GOROOT` to find them.

deadprogram, 2021-12-11:
#2376 has been merged into the `dev` branch. We will close this issue after the next release. Thank you!

closed 2022-01-26 by deadprogram