tinygo-org/tinygo #2377: USB flag for arduino uno not supported

jadefox10200, 2021-12-10:
I am trying to do a println to the console. I wanted to get this working for debugging purposes. I assume I am doing something wrong as I have looked everywhere but can't find a resolution and have tried this on two different computers with the same error. 

The following is my program:
```package main

import (
	"time"
)

func main() {
	for {
		println("Hello")
		time.Sleep(time.Second)
	}
}```

I am typing this into the console: 
```tinygo flash -serial usb -target arduino -port /dev/ttyACM0 main.go```

I am getting the following error: 
``/usr/local/lib/tinygo/src/machine/serial-usb.go:6:14: undeclared name: USB```

Per the help file this is a valid flag but it doesn't seem to be available for the Arduino Uno.

aykevl, 2021-12-11:
`-serial=usb` is only supported on devices that use USB-CDC. The Arduino Uno has a simple serial port, that is used both for flashing and for monitoring. You can simply leave out the flag (or set it to `-serial=uart`) and `println` output will appear in the serial console (9600 baud).

aykevl, 2021-12-11:
In general, TinyGo will try to use sane defaults. This should also work:

```
tinygo flash -target arduino main.go
```

jadefox10200, 2021-12-12:
> In general, TinyGo will try to use sane defaults. This should also work:
> 
> ```
> tinygo flash -target arduino main.go
> ```

I have tried this previously as the first option, however there is nothing in the console when I run this command. I do not get any feedback from the program upon flashing it. When the flash completes, the terminal is "free" meaning that the program is not running nor communicating with the console. I do not quite understand where Stdout is writing to.

jadefox10200, 2021-12-13:
Ok, I get it now. I did not realize I had to run the Serial monitor from the Arduino IDE. I thought I could get the output directly back to the terminal console that I was working in.

deadprogram, 2021-12-13:
You just have to run some serial connection from your computer to the Arduino. It could be `picocom` or `minicom` or could be the Arduino IDE monitor. The point is that some serial program needs to connect from the computer to the board since the code you have just uploaded is executing on the board itself.

jadefox10200, 2021-12-16:
Yes, I understand my issue now. Thank you for your assistance and apologize for the confusion.

closed 2021-12-16 by jadefox10200

eerison, 2023-04-16:
> output will appear in the serial console (9600 baud)

do you have some example, how to do this?