tinygo-org/tinygo #2382: google/open-location-code/go tests fail on wasi due to asynchronous file loading

dgryski, 2021-12-13:
This `init()` function here https://github.com/google/open-location-code/blob/main/go/olc_test.go#L63 loads some test files asynchronously.  On wasi, this causes the test to fail opening the file.  Changing the init function to load them in sequence allows the test to pass.

dgryski, 2021-12-13:
cc @niaow for scheduler bugs

dgryski, 2021-12-18:
I wonder if we're `wasi-libc` is insufficiently thread-safe for how we're using it?

niaow, 2021-12-18:
What is the failure though?

dgryski, 2021-12-18:
```

niaow, 2021-12-22:
This bug appears to be caused or triggered by interp.

niaow, 2021-12-22:
The actual issue here is that interp is placing these on the runqueue, which causes them to run before the inits are run, causing things to break horribly.

dgryski, 2021-12-24:
This test now passes.

```
dgryski@beekin go % ~/go/src/github.com/tinygo-org/tinygo/build/tinygo test -c -target=wasi
dgryski@beekin go % wasmtime run ./go.test --dir=. --dir=.. -- -test.v
=== RUN   TestCheck
--- PASS: TestCheck
=== RUN   TestEncode
--- PASS: TestEncode
=== RUN   TestDecode
--- PASS: TestDecode
=== RUN   TestShorten
--- PASS: TestShorten
=== RUN   TestFuzzCrashers
--- PASS: TestFuzzCrashers
PASS
```

closed 2022-01-26 by deadprogram