tinygo-org/tinygo #2399: compiler: add support for 'any' as alias for interface{}

dolmen, 2021-12-17:
Add support for `any` as an built-in alias for type `interface{}`, as this will appear in Go 1.18.

References:
* https://github.com/golang/go/issues/33232#issuecomment-915333205
* https://tip.golang.org/ref/spec#Predeclared_identifiers

aykevl, 2021-12-17:
This is probably be added automatically once we add support for Go 1.18. I don't think we need to do anything in TinyGo, and if we do, that should be tested after we add support for Go 1.18.

dolmen, 2021-12-20:
Does that mean that the TinyGo compiler is closely in sync with the main Go compiler or even shares code? I lack knowledge about the compiler architecture of TinyGo. I expected to find information in the "[Compiler Internals](https://tinygo.org/docs/concepts/compiler-internals/)" section of the documentation, but I see nothing there about that.

aykevl, 2021-12-28:
The page you're looking for is this one: https://tinygo.org/docs/concepts/compiler-internals/pipeline/

> * Lexing, parsing, and typechecking is done by packages in the standard library and in the golang.org/x/tools/go library.
> * SSA construction (a very important step) is done by the golang.org/x/tools/go/ssa package.

So basically, all syntax and types are handles by the Go standard library.

closed 2022-02-01 by dolmen