tinygo-org/tinygo #2400: wasmtest: TestChan failing randomly in CI

niaow, 2021-12-17:
It seems that TestChan is failing occasionally in CI under `assert-test-linux` due to a timeout.
This could just be a timeout (10 seconds does seem like it should be plenty though) or it could be a bug.

niaow, 2021-12-17:
I am somewhat unsure of what is going on. No approach I try seems to be able to get this test to take over 250 ms (running it pinned to a single core to try to CPU-starve it, running it repeatedly, etc.).

zdima, 2021-12-27:
I have PR #2431 (draft) that I believe will address this issue. 
Is there a way to trigger CI without making additional commits or better run “reduced” test on CI environment with “make wasmtest” only?

deadprogram, 2021-12-27:
> Is there a way to trigger CI without making additional commits

I can help you with that. I will retrigger this right now as another test. Also please see my comments.

dkegel-fastly, 2022-01-18:
This is happenig a lot still.

deadprogram, 2024-05-12:
I think this was addressed some time ago. Now closing, please reopen if needed. Thanks everyone!

closed 2024-05-12 by deadprogram