tinygo-org/tinygo #2402: out of bounds memory access during gonum's TestDgeev test under wasi

dgryski, 2021-12-18:
```
~/go/src/github.com/dgryski/tinygo-test-corpus/_corpus/gonum/gonum/lapack/gonum $ WASMTIME_BACKTRACE_DETAILS=1 tinygo test -tags=noasm -target=wasi -v -run=TestDgeev
=== RUN   TestDgeev
Error: failed to run main module `/var/folders/5b/_hr1d1fd3qn4t9vtfx5p61wm0000gp/T/tinygo1370638171/main`

Caused by:
    0: failed to invoke command default
    1: wasm trap: out of bounds memory access
       wasm backtrace:
           0: 0x8eef - (runtime.gcBlock).state
                           at /Users/dgryski/go/src/github.com/tinygo-org/tinygo/src/runtime/gc_conservative.go:146:20
           1: 0x3f38 - runtime.alloc
                           at /Users/dgryski/go/src/github.com/tinygo-org/tinygo/src/runtime/gc_conservative.go:312:17
           2: 0x1601f - runtime.stringFromUnicode
                           at /Users/dgryski/go/src/github.com/tinygo-org/tinygo/src/runtime/string.go:133:2
           3: 0xa841e - (gonum.org/v1/gonum/lapack/gonum.Implementation).Dhseqr
                           at /Users/dgryski/go/src/github.com/dgryski/tinygo-test-corpus/_corpus/gonum/gonum/lapack/gonum/dhseqr.go:207:42
           4: 0x7ff99 - <unknown>!(gonum.org/v1/gonum/lapack/gonum.Implementation).Dgeev
           5: 0x7cf15 - (gonum.org/v1/gonum/lapack/gonum.Implementation).Dgeev$invoke
                           at /Users/dgryski/go/src/github.com/dgryski/tinygo-test-corpus/_corpus/gonum/gonum/lapack/gonum/dgeev.go:63:28
                     - (Go interface method)
                           at /<Go interface method>
                     - gonum.org/v1/gonum/lapack/testlapack.testDgeev
                           at /Users/dgryski/go/src/github.com/dgryski/tinygo-test-corpus/_corpus/gonum/gonum/lapack/testlapack/dgeev.go:590:21
           6: 0x77b2a - gonum.org/v1/gonum/lapack/testlapack.DgeevTest

           7: 0x1371ec - gonum.org/v1/gonum/lapack/gonum.TestDgeev
                           at /Users/dgryski/go/src/github.com/dgryski/tinygo-test-corpus/_corpus/gonum/gonum/lapack/gonum/lapack_test.go:52:22
           8: 0x12c90 - flag.Parse
                           at /Users/dgryski/go/src/github.com/tinygo-org/tinygo/src/testing/testing.go:293:9
                     - (*testing.M).Run
                           at /Users/dgryski/go/src/github.com/tinygo-org/tinygo/src/testing/testing.go:255:13
                     - main.main
                           at /Users/dgryski/Library/Caches/go-build/6c/6c9b580934e9b6b63336327ecd58d4e4726db7d194f1fff3bfcbc43be9f99c0f-d:299:15
                     - runtime.run$1
                           at /Users/dgryski/go/src/github.com/tinygo-org/tinygo/src/runtime/scheduler_any.go:26:11
           9: 0x108b5 - <goroutine wrapper>
                           at /Users/dgryski/go/src/github.com/tinygo-org/tinygo/src/runtime/scheduler_any.go:23:2
          10: 0x1c218b - <unknown>!tinygo_launch
          11: 0x3638 - (*internal/task.Task).Resume
                           at /Users/dgryski/go/src/github.com/tinygo-org/tinygo/src/internal/task/task_asyncify.go:110:17
          12: 0x107ef - _start
                           at /Users/dgryski/go/src/github.com/tinygo-org/tinygo/src/runtime/scheduler.go:171:11

FAIL	gonum.org/v1/gonum/lapack/gonum	0.208s
FAIL
```

dgryski, 2021-12-18:
This test passes with `-gc=leaking`.

dgryski, 2021-12-22:
This is now failing with a wasi stack overflow:
```
dgryski@Damians-MBP-2 gonum % ~/go/src/github.com/tinygo-org/tinygo/build/tinygo test -target=wasi -c
dgryski@Damians-MBP-2 gonum % WASMTIME_BACKTRACE_DETAILS=1 wasmtime run ./gonum.test -- -test.run=TestDgeev
panic: runtime error: stack overflow
Error: failed to run main module `./gonum.test`
```

which means this can probably be fixed with a custom target.json.

niaow, 2021-12-30:
Is #2447 enough for this?

dgryski, 2021-12-30:
Yup, passes now.

```
~/go/src/github.com/dgryski/tinygo-test-corpus/_corpus/gonum/gonum/lapack/gonum $  ~/go/src/github.com/tinygo-org/tinygo/build/tinygo test -target=wasi -run=TestDgeev
ok  	gonum.org/v1/gonum/lapack/gonum	42.976s
```

closed 2022-01-26 by deadprogram