tinygo-org/tinygo #2404: tinygo should support passing other arguments to the emulator

dgryski, 2021-12-18:
A number of tests require more filesystem access that simply the current directory.  Being able to pass additional `--dir=` flags to wasmtime would allow `tinygo test -target=wasi` to pass more package tests.

deadprogram, 2022-04-28:
I think this was addressed by https://github.com/tinygo-org/tinygo/pull/2780

dgryski, 2022-09-12:
```

dgryski, 2022-09-14:
`--extra-emu-flags=...`

codefromthecrypt, 2023-01-12:
I saw this while porting wazero to work with things like this. FYI the wasi calls look like the below. Nevermind the NOENT as that's a thing I'm working on.. the important thing are the inputs

```bash
$ ./build/tinygo test -target wasi -c -o flate.wasm compress/flate
$ wazero run -hostlogging=filesystem -mount=.:/  flate.wasm -test.v
==> wasi_snapshot_preview1.fd_prestat_get(fd=3)
<== (prestat={pr_name_len=1},errno=ESUCCESS)
==> wasi_snapshot_preview1.fd_prestat_dir_name(fd=3)
<== (path=/,errno=ESUCCESS)
==> wasi_snapshot_preview1.fd_prestat_get(fd=4)
<== (prestat=,errno=EBADF)
==> wasi_snapshot_preview1.fd_fdstat_get(fd=3)
<== (stat={filetype=DIRECTORY,fdflags=,fs_rights_base=,fs_rights_inheriting=},errno=ESUCCESS)
==> wasi_snapshot_preview1.path_open(fd=3,dirflags=SYMLINK_FOLLOW,path=../testdata/e.txt,oflags=,fs_rights_base=,fs_rights_inheriting=,fdflags=)
<== (opened_fd=,errno=ENOENT)
==> wasi_snapshot_preview1.fd_fdstat_get(fd=3)
<== (stat={filetype=DIRECTORY,fdflags=,fs_rights_base=,fs_rights_inheriting=},errno=ESUCCESS)
==> wasi_snapshot_preview1.path_open(fd=3,dirflags=SYMLINK_FOLLOW,path=../../testdata/Isaac.Newton-Opticks.txt,oflags=,fs_rights_base=,fs_rights_inheriting=,fdflags=)
<== (opened_fd=,errno=ENOENT)
==> wasi_snapshot_preview1.fd_fdstat_get(fd=3)
<== (stat={filetype=DIRECTORY,fdflags=,fs_rights_base=,fs_rights_inheriting=},errno=ESUCCESS)
==> wasi_snapshot_preview1.path_open(fd=3,dirflags=SYMLINK_FOLLOW,path=../../testdata/Isaac.Newton-Opticks.txt,oflags=,fs_rights_base=,fs_rights_inheriting=,fdflags=)
<== (opened_fd=,errno=ENOENT)
```

dgryski, 2023-01-12:
Yeah, there was some tricks involved to get all the proper directories added for testing `compress/flate`: https://github.com/tinygo-org/tinygo/blob/release/main.go#L254