tinygo-org/tinygo #2409: interp: wait loops run forever

niaow, 2021-12-19:
When investigating some failures in interp, I came up with this test case:
```
package main

import(
        "sync/atomic"
        "time"
)

func init() {
        var x uint32
        go func() {
                atomic.StoreUint32(&x, 1)
        }()
        for atomic.LoadUint32(&x) != 1 {
                time.Sleep(time.Millisecond)
        }
}

func main() {
}
```

I think the correct behavior here would be to pass the loop through as is.
Additionally, if the loop contains a lock (or similar) and the bounds are known, then interp will effectively be unrolling the loop (which is not great if the loop runs millions of times).

niaow, 2021-12-19:
I think I am mixing things up. I am very confused right now.

aykevl, 2021-12-28:
This code contains a branch based on a value that's only known at runtime, so it _should_ bail out and run the whole function at runtime. These lines are somewhat relevant:

https://github.com/tinygo-org/tinygo/blob/9fa667ce6341fe653f2cac571dd0a262fb390cdf/interp/interpreter.go#L100-L101

Basically, if any of the instruction operands (in this case, a `llvm.CondBr`) is not known, it attempts to run that instruction at runtime.
However, branch instructions can't easily be run at runtime, so abort the whole function:

https://github.com/tinygo-org/tinygo/blob/9fa667ce6341fe653f2cac571dd0a262fb390cdf/interp/interpreter.go#L987-L988

I originally attempted to make branch instructions work, but seeing this I think it's a good idea to always bail and run the function at runtime. We absolutely don't want loops like this to be unrolled.

niaow, 2021-12-28:
> I originally attempted to make branch instructions work, but seeing this I think it's a good idea to always bail and run the function at runtime. We absolutely don't want loops like this to be unrolled.

Yeah. 

I need to look back at this later, but I suspect that the atomic operation in this specific example is being run at interp time, assuming (incorrectly) that `x` could not have been modified by anything else.

davecheney, 2026-10-26:
The example now compiles and completes with TinyGo 0.44-dev (`bcbb86e5`) on macOS arm64. The goroutine stores the flag, the initialization loop exits, and `main` runs. Closing this as resolved. Please reopen if you can still reproduce it with a recent TinyGo version.

closed 2026-10-26 by davecheney