tinygo-org/tinygo #2415: standalone test binaries should print `PASS` or `FAIL` when finished

dgryski, 2021-12-20:
```
~/go/src/go.googlesource.com/go/src/compress/flate $ tinygo test -c compress/flate && ./flate.test
~/go/src/go.googlesource.com/go/src/compress/flate $ go test -c &&  ./flate.test
PASS
```

aykevl, 2021-12-28:
Huh, that must have been a bug I've introduced here: https://github.com/tinygo-org/tinygo/pull/2043
So essentially what we need is that test binaries print "PASS" or "FAIL" but if they pass, they shouldn't print this (to avoid excessive output, like before #2043).

BerndCzech, 2022-12-26:
Can it be closed as #3229 was merged? Or can I help here?

dgryski, 2022-12-26:
It should be tagged as "next release". Bugs are usually closed when they are merged into the release branch, not just the development version.

deadprogram, 2023-02-12:
0.27.0 has now been released, so closing. Thank you!

closed 2023-02-12 by deadprogram