tinygo-org/tinygo #2420: tinygo build failed on M1 macOS

PureWhiteWu, 2021-12-22:
Because homebrew has changed it's directory to `/opt/homebrew` instead of `/usr/local` on  Apple Silicon macOS, so compile will fail with "file not found" error.

aykevl, 2021-12-28:
I don't have a macOS, and I certainly don't have one of those shiny new M1 systems. However, I think you can help me fix this.

Can you do the following?

- Clone go-llvm in a separate directory: `git clone https://github.com/tinygo-org/go-llvm.git`
- Modify go.mod in the TinyGo root directory to add `replace tinygo.org/x/go-llvm => /path/to/your/go-llvm` at the end
- Apply the following patch to go-llvm that you just cloned:

```diff
diff --git a/llvm_config_darwin.go b/llvm_config_darwin.go
index 7882d08..530ecc8 100644
--- a/llvm_config_darwin.go
+++ b/llvm_config_darwin.go
@@ -4,9 +4,12 @@ package llvm
 
 // Automatically generated by `make config BUILDDIR=`, do not edit.
 
-// #cgo CPPFLAGS: -I/usr/local/opt/llvm@11/include   -D__STDC_CONSTANT_MACROS -D__STDC_FORMAT_MACROS -D__STDC_LIMIT_MACROS
-// #cgo CXXFLAGS: -std=c++14
-// #cgo LDFLAGS: -L/usr/local/opt/llvm@11/lib -Wl,-search_paths_first -Wl,-headerpad_max_install_names -lLLVM -lz -lcurses -lm -lxml2 -L/usr/local/opt/libffi/lib -lffi
+// #cgo amd64 CPPFLAGS: -I/usr/local/opt/llvm@11/include   -D__STDC_CONSTANT_MACROS -D__STDC_FORMAT_MACROS -D__STDC_LIMIT_MACROS
+// #cgo amd64 CXXFLAGS: -std=c++14
+// #cgo amd64 LDFLAGS: -L/usr/local/opt/llvm@11/lib -Wl,-search_paths_first -Wl,-headerpad_max_install_names -lLLVM -lz -lcurses -lm -lxml2 -L/usr/local/opt/libffi/lib -lffi
+// #cgo arm64 CPPFLAGS: -I/opt/homebrew/llvm@11/include   -D__STDC_CONSTANT_MACROS -D__STDC_FORMAT_MACROS -D__STDC_LIMIT_MACROS
+// #cgo arm64 CXXFLAGS: -std=c++14
+// #cgo arm64 LDFLAGS: -L/opt/homebrew/llvm@11/lib -Wl,-search_paths_first -Wl,-headerpad_max_install_names -lLLVM -lz -lcurses -lm -lxml2 -L/opt/homebrew/libffi/lib -lffi
 import "C"
 
 type run_build_sh int
```

And then try to build it again. Note that you need to have `llvm@11` installed for this to work.

You probably need to make a similar patch to tinygo/cgo/libclang_config.go:

```diff
diff --git a/cgo/libclang_config.go b/cgo/libclang_config.go
index 9f7cdc1c..047aaae3 100644
--- a/cgo/libclang_config.go
+++ b/cgo/libclang_config.go
@@ -5,10 +5,12 @@ package cgo
 
 /*
 #cgo linux   CFLAGS:  -I/usr/lib/llvm-11/include
-#cgo darwin  CFLAGS:  -I/usr/local/opt/llvm@11/include
+#cgo darwin amd64 CFLAGS:  -I/usr/local/opt/llvm@11/include
+#cgo darwin ard64 CFLAGS:  -I/opt/homebrew/llvm@11/include
 #cgo freebsd CFLAGS:  -I/usr/local/llvm11/include
 #cgo linux   LDFLAGS: -L/usr/lib/llvm-11/lib -lclang
-#cgo darwin  LDFLAGS: -L/usr/local/opt/llvm@11/lib -lclang -lffi
+#cgo darwin amd64 LDFLAGS: -L/usr/local/opt/llvm@11/lib -lclang -lffi
+#cgo darwin arm64 LDFLAGS: -L/opt/homebrew/llvm@11/lib -lclang -lffi
 #cgo freebsd LDFLAGS: -L/usr/local/llvm11/lib -lclang
 */
 import "C"
```

If this works, it would be great if you can make a PR to go-llvm and to TinyGo with these changes.

PureWhiteWu, 2021-12-29:
Thanks for your reply! I've already solved this issue locally before, and I will submit two PRs to fix this.

PureWhiteWu, 2021-12-29:
Hi, I've submitted two PRs https://github.com/tinygo-org/go-llvm/pull/27 and https://github.com/tinygo-org/tinygo/pull/2439.
There are still two problems in Makefile that I'm not quite sure how to fix at this time, so I left two comments there.

PureWhiteWu, 2021-12-30:
Hi, there's another PR needs review and merge: https://github.com/tinygo-org/tinygo/pull/2439.

PR #2439 (2022-01-12): feat: support build on darwin arm64
fixes #2420
closed 2022-01-26 by deadprogram