tinygo-org/tinygo #2423: time.Sleep doesn't work as expected on Arduino Nano

cweagans, 2021-12-23:
I'm doing some testing with an Arduino Nano and it looks like `time.Sleep` doesn't do the right thing. Here's the code I'm using to test:

```go
package main

import (
	"machine"
	"time"
)

const delay = 16

func main() {
	pin := machine.D12
	pin.Configure(machine.PinConfig{Mode: machine.PinOutput})
	for {
		pin.High()
		time.Sleep(time.Millisecond * delay)
		pin.Low()
		time.Sleep(time.Millisecond * delay)
	}
}
```

When I change `delay` to different powers of 2, here is the timing of each iteration of the for loop (as measured by my logic analyzer):

* `2` : `5 us`
* `4` : `5 us`
* `8` : `5 us`
* `16`: `5 us`
* `32`: `33ms`
* `64`: `101ms`
* `128`: `236ms`
* `256`: `506ms`
* `512`: `1.01s` 

I would definitely expect a little variance given that Arduinos aren't exactly high precision devices, but this seems pretty unpredictable.

Compare to the following Arduino sketch:

```c
void setup() {
  pinMode(12, OUTPUT);
}

void loop() {
  const int wait = 2;
  digitalWrite(12, HIGH);
  delay(wait);
  digitalWrite(12, LOW);
  delay(wait);
}
```

Modifying `wait` in the same way as before:

* `2` : `2 ms`
* `4` : `8 ms`
* `8` : `16 ms`
* `16`: `32 ms`
* `32`: `63.9 ms`
* `64`: `127.9ms`
* `128`: `255.9ms`
* `256`: `511.7ms`
* `512`: `1s` 

It seems like mostly accurate timing is doable, but I don't understand enough about tinygo under the hood to know why the behavior is different. I saw a couple of issues open for the Arduino Uno (which also has the atmega328p), so maybe it's something to do with the support for this specific mcu?

While I'd like to get this resolved, I'm more interested in getting up to speed with TinyGo. Can you help me get pointed in the right direction so that I can debug and fix this myself? I'm still coming up to speed with everything, but my goal is to work on TinyGo more regularly. Understanding (hopefully) little problems like this help a ton!

ysoldak, 2021-12-23:
Seem to be very MCU dependent.
I've verified your sketch on couple of my boards:

Nano 33 BLE (nrf52840)
- `4`:  `8ms`
- `128`: `256ms`
Perfect!

Nano RP2040 (RP2040)
- `4`:  `8ms` and sometimes `8.04ms`
- `128`: `256ms` and sometimes `256.04ms`
Almost perfect with odd `0.04ms` added sometimes

Seeeduino XIAO (SAMD21G18)
- `4`:  `8.52ms`
- `128`: `257.8ms`
Basically adds `0.2 - 0.9 ms` per sleep call depending on the delay value

aykevl, 2021-12-23:
This is a duplicate of https://github.com/tinygo-org/tinygo/issues/952. Please see that issue for more details.

Here is the relevant code for the ATmega328p (Arduino Uno/Nano):
https://github.com/tinygo-org/tinygo/blob/06df31944c6af706d6cb8b30a2e92d19060436d2/src/runtime/runtime_atmega.go#L18-L24

soypat, 2021-12-23:
This has bit me several times and impeded in my development of the ENC28J60 driver and ethernet stack unmarshaller. I'd also really like to see it fixed though I'm currently swamped with work until sometime this January.

cweagans, 2021-12-23:
Okay, so I've read through some of the other related issues. It seems like there's some question about whether or not to take one of the timers over for time keeping (which has the result of messing with PWM).

Is it okay to take over one of the timers to fix this? Happy to dive in and work on it, but I'd like to make sure I do it in a way that will be acceptable in tinygo.

zdima, 2021-12-24:
@cweagans, I don't have tools to measure timing. Can you try interrupt based implementation of time and Sleep for AVR? 
This is my branch (based on latest dev branch): https://github.com/zdima/tinygo/tree/avr-timer

cweagans, 2021-12-24:
Certainly. I'll see if I can get to it tonight, but it'll be Sunday otherwise.

cweagans, 2021-12-24:
@zdima Your branch is a _huge_ improvement over the original results. Here are the timings I recorded with the same code as the main issue body.

* `2` : `4 ms`
* `4` : `8 ms`
* `8` : `16 ms`
* `16`: `32 ms`
* `32`: `63.9ms`
* `64`: `127.9ms`
* `128`: `255.8ms`
* `256`: `511.7ms`
* `512`: `1s`

cweagans, 2022-01-02:
Fixed by https://github.com/tinygo-org/tinygo/pull/2428

closed 2022-01-02 by cweagans