tinygo-org/tinygo #2424: board support

markgardner10, 2021-12-23:
I would like to start with tinygo but not sure if any of my existing boards will work

I currently have the following boards

STM32F411 Nucleo-64 card

STM32F402 Discovery kit

STM32F411 Discovery kit

TIVA C Series TM4C123G Launchpad Evaluation kit

Raspberry pi zero

Rasperbby pi v3 8gb ram

Waveshare beaglebone black rev c 1ghz ARM cortex-a8 512mb ddr3 4gb 8bit emmc board - mini pc development board

Any advice appreciated

Please suggest a suitable good all round board if none of the above suffice

Thanks

niaow, 2021-12-23:
> Raspberry pi zero
> Raspberry pi v3 8gb ram
> Waveshare beaglebone black rev c 1ghz ARM cortex-a8 512mb ddr3 4gb 8bit emmc board - mini pc development board

These three should all "work" in the sense that you can just build regular Linux binaries for them by passing `GOARCH=arm` (or `GOARCH=arm64` if you are targeting a 64-bit OS) to `tinygo` as an environment variable. If you want to compile on the board itself, we also have 32-bit ARM packages for the compiler with our releases.

> STM32F411 Nucleo-64 card
> STM32F402 Discovery kit
> STM32F411 Discovery kit

It does not look like we currently support any of these particular STM32 boards. We currently support the following STM32 boards (copied from the readme):
- ST Micro "Nucleo" F103RB
- ST Micro "Nucleo" F722ZE
- ST Micro "Nucleo" L031K6
- ST Micro "Nucleo" L432KC
- ST Micro "Nucleo" L552ZE
- ST Micro "Nucleo" WL55JC
- ST Micro STM32F103XX "Bluepill"
- ST Micro STM32F407 "Discovery"

If you wanted to add any of the STM32 boards you listed, it would probably not be terribly difficult. https://github.com/tinygo-org/tinygo/wiki/Adding-a-new-board

I currently mostly use an Arduino Nano 33 IoT and a Metro M4 Express Airlift, both of which have WiFi support through https://github.com/tinygo-org/drivers

markgardner10, 2021-12-24:
thanks for the info - appreciate it @niaow

deadprogram, 2024-11-27:
Closing since this question appears to have been answered. Thanks!

closed 2024-11-27 by deadprogram