tinygo-org/tinygo #2432: Environ not declared by package syscall

Inkeliz, 2021-12-28:
I'm trying to test `tinygo build -o wasm.wasm -target wasm .` on `tinygo version 0.21.0 windows/amd64 (using go version go1.17.1 and LLVM version 11.0.0)`, but gives:

```
# internal/syscall/execenv
C:\Program Files\Go\src\internal\syscall\execenv\execenv_default.go:19:17: Environ not declared by package syscall
```

Inkeliz, 2021-12-28:
That was cause by `os.Exec`, that was used on `Windows` and not proper set `+build !js`. However, I expect that `os.Exec` act as `no-op`.

niaow, 2021-12-28:
Right now, TinyGo does not support `os/exec` at all yet. It is one of the more complex packages to support, since it depends on a lot of functionality which has not been implemented yet (e.g. concurrent I/O). It might be reasonable to add a not supported stub for now.

niaow, 2021-12-28:
Related issue #632

closed 2021-12-28 by Inkeliz