tinygo-org/tinygo #2438: Makefile doesn't support macOS now

PureWhiteWu, 2021-12-29:
LLVM auto detect doesn't work on macOS now, because homebrew installed llvm is key-only:
```
$ brew info llvm
llvm: stable 13.0.0 (bottled), HEAD [keg-only]
...
==> Caveats
To use the bundled libc++ please add the following LDFLAGS:
  LDFLAGS="-L/opt/homebrew/opt/llvm/lib -Wl,-rpath,/opt/homebrew/opt/llvm/lib"

llvm is keg-only, which means it was not symlinked into /opt/homebrew,
because macOS already provides this software and installing another version in
parallel can cause all kinds of trouble.
```

To workaround this, we need to modify this `Makefile` and manually specify the correct path for llvm on mac. 
For arm64(M1):
```
CLANG ?= /opt/homebrew/Cellar/llvm@11/11.1.0_3/bin/clang
LLVM_AR ?= /opt/homebrew/Cellar/llvm@11/11.1.0_3/bin/llvm-ar
LLVM_NM ?= /opt/homebrew/Cellar/llvm@11/11.1.0_3/bin/llvm-nm
```

For amd64(x86_64):
```
CLANG ?= /usr/local/Cellar/llvm@11/11.1.0_3/bin/clang
LLVM_AR ?= /usr/local/Cellar/llvm@11/11.1.0_3/bin/llvm-ar
LLVM_NM ?= /usr/local/Cellar/llvm@11/11.1.0_3/bin/llvm-nm
```

niaow, 2021-12-31:
If I understand correctly, this should be using XCode's clang (which we seem to use in CI).

PureWhiteWu, 2022-01-01:
> If I understand correctly, this should be using XCode's clang (which we seem to use in CI), not Homebrew's. This doesn't seem to be documented though.

But using Xcode's clang won't compile.

deadprogram, 2022-01-11:
> If I understand correctly, this should be using XCode's clang (which we seem to use in CI), not Homebrew's. This doesn't seem to be documented though.

No, we only work with either the homebrew install, or else our own custom build of LLVM (used to include the Xtensa bits). As @PureWhiteWu points out, it does not work with the XCode clang.

niaow, 2022-01-11:
Oh, this is for building wasi-libc.

PureWhiteWu, 2022-01-12:
> Oh, this is for building wasi-libc.

wasi-libc also won't compile with Xcode's clang.

deadprogram, 2022-01-14:
@PureWhiteWu are you able to submit a PR with a fix for this? 

I think `$(brew --prefix)` will return the correct prefix on either platform.

PureWhiteWu, 2022-01-17:
> @PureWhiteWu are you able to submit a PR with a fix for this?
> 
> I think `$(brew --prefix)` will return the correct prefix on either platform.

Thanks for instruction!
I've submitted an PR #2535 which should fix this.
I've tested it locally and it works in my environment(M1 macOS), but I didn't test it on Intel macOS.

dkegel-fastly, 2022-01-17:
What are the commands to reproduce the problem?  I have a mac and can test, but the interplay between brew and tinygo has always mystified me.

deadprogram, 2022-01-17:
If you are trying to use `make` to build TinyGo from source on macOS, and you are also using the Homebrew install of LLVM (not the "byollvm" option from running `make llvm-source && make llvm-build`), there was a problem due to changes in how homebrew installs LLVM now as "keg-only".

The PR fixes that, and at the same time makes it work on both Intel and M1 Mac hardware.

dkegel-fastly, 2022-01-17:
Well, yes, but what are the commands for that?  https://tinygo.org/docs/guides/build/ is a bit confusing.
I don't think I'm even getting to the command this is trying to get working, whatever it is.

When I try
```
brew install llvm@13
go install
```
I get the error
```
fatal error: 'clang-c/Index.h' file not found
```

If I try 
```
make
```
it fails with
```
Fetch and build LLVM first by running:
  make llvm-source
  make llvm-build
```

PureWhiteWu, 2022-01-18:
The command should be `make wasi-libc`

PR #2535 (2022-01-17): fix: makefile on macOS
fixes #2438.
closed 2022-01-26 by deadprogram