tinygo-org/tinygo #2441: x509.ParsePKCS1PublicKey() failes due to panic: unimplemented: (reflect.Type).Name()

stellirin, 2021-12-29:
Envoy crashes when parsing a PEM formatted public key with the following:

```
[source/extensions/common/wasm/context.cc:1172] wasm log: panic: unimplemented: (reflect.Type).Name()
[source/extensions/common/wasm/wasm_vm.cc:38] Function: proxy_on_context_create failed: Uncaught RuntimeError: unreachable
```

I am using the latest development containers of TinyGo and Envoy (I didn't try it on any other VM):

```
envoyproxy/envoy-dev:latest
tinygo/tinygo-dev:latest
```

I created a 'minimal' variant of the code to reproduce in Envoy:

```go
package main

import (
	"encoding/asn1"
	"math/big"

	"github.com/tetratelabs/proxy-wasm-go-sdk/proxywasm"
	"github.com/tetratelabs/proxy-wasm-go-sdk/proxywasm/types"
)

const der = `-----BEGIN RSA PUBLIC KEY-----
MIIBCgKCAQEAyZyOUhd20bTFly59xEeIKTp0weUHcbsV8FDBBWQXnA81yah3rJqa
hHYFKPmXeTVNjjD2yPgQpMTE6I4BAc/rjG7kiC5mG1A84WjH2GsWnJ7CRE/E1lGL
qjBP4tLkYe0IJWv2dJtYS5leierPjvdon6qI5le/EWg9gSimJX/VrxM9BlHCH3S1
BPONX2nsD5PHVH1Ad8toPI4orz0I1psRj5EFDL5UeIqjiWwwHK/Z32jXFZMLDIrB
ACPvkaBYW4gBywogPTGL5C7uamwQc+RJxWWVC7eGDqQlZB9HjwTOxH1zq2nHlHRg
HFYoVW7QedsxgvgqWeuvLyhLEd3wsXXMAQIDAQAB
-----END RSA PUBLIC KEY-----`

type vmContext struct {
	types.DefaultVMContext

	// publicKey reflects the ASN.1 structure of a PKCS #1 public key.
	publicKey struct {
		N *big.Int
		E int
	}
}

func (ctx *vmContext) NewPluginContext(_ uint32) types.PluginContext {
	asn1.UnmarshalWithParams([]byte(der), &ctx.publicKey, "")
	return &types.DefaultPluginContext{}
}

func main() {
	proxywasm.SetVMContext(&vmContext{})
}
```

I build the filter with:

```sh
docker run -it --rm \
  --volume=/workspace \
  --workdir=/workspace \
  tinygo/tinygo-dev:latest \
        tinygo \
        build \
        -opt=2 \
        -no-debug \
        -target=wasi \
        -scheduler=none \
        -o=test.wasm .
```

dgryski, 2021-12-29:
The `encoding/asn1` package needs parts of `reflect` that aren't yet implemented. A smaller test is just to try to run the `encoding/asn1` package tests.

SamYuan1990, 2022-01-01:
so which means if we can't use x509 related packages for now?
and where can we find a guide line to make a go lang package able to be packaged by tinygo?

deadprogram, 2022-01-19:
See https://github.com/tinygo-org/tinygo/pull/2479

dgryski, 2023-02-24:
The following code extracted from your example produces the same error in tinygo and mainline go:
```
package main

import (
        "encoding/asn1"
        "fmt"
        "math/big"
)

const der = `-----BEGIN RSA PUBLIC KEY-----
MIIBCgKCAQEAyZyOUhd20bTFly59xEeIKTp0weUHcbsV8FDBBWQXnA81yah3rJqa
hHYFKPmXeTVNjjD2yPgQpMTE6I4BAc/rjG7kiC5mG1A84WjH2GsWnJ7CRE/E1lGL
qjBP4tLkYe0IJWv2dJtYS5leierPjvdon6qI5le/EWg9gSimJX/VrxM9BlHCH3S1
BPONX2nsD5PHVH1Ad8toPI4orz0I1psRj5EFDL5UeIqjiWwwHK/Z32jXFZMLDIrB
ACPvkaBYW4gBywogPTGL5C7uamwQc+RJxWWVC7eGDqQlZB9HjwTOxH1zq2nHlHRg
HFYoVW7QedsxgvgqWeuvLyhLEd3wsXXMAQIDAQAB
-----END RSA PUBLIC KEY-----`

type publicKey struct {
        N *big.Int
        E int
}

func main() {
        var pub publicKey
        _, err := asn1.UnmarshalWithParams([]byte(der), &pub, "")
        if err != nil {
                fmt.Println(err)
                return
        }
        fmt.Println(pub.N)
}
```

```
asn1: structure error: tags don't match (16 vs {class:0 tag:13 length:45 isCompound:true}) {optional:false explicit:false application:false private:false defaultValue:<nil> tag:<nil> stringType:0 timeType:0 set:false omitEmpty:false} publicKey @2
~/go/src/github.com/dgryski/bug/arrref $ tinygo run main.go
asn1: structure error: tags don't match (16 vs {class:0 tag:13 length:45 isCompound:true}) {optional:false explicit:false application:false private:false defaultValue:<nil> tag:<nil> stringType:0 timeType:0 set:false omitEmpty:false} publicKey @2
```

I believe this issue is fixed by #3470 .

codefromthecrypt, 2023-06-12:
I can compile and run @dgryski's example. compile with TinyGo v0.28.1 and run with wazero 1.2.0. So, is this closeable?

```bash
$ cat main.go
cat main.go 
package main

import (
	"encoding/asn1"
	"fmt"
	"math/big"
)

const der = `-----BEGIN RSA PUBLIC KEY-----
MIIBCgKCAQEAyZyOUhd20bTFly59xEeIKTp0weUHcbsV8FDBBWQXnA81yah3rJqa
hHYFKPmXeTVNjjD2yPgQpMTE6I4BAc/rjG7kiC5mG1A84WjH2GsWnJ7CRE/E1lGL
qjBP4tLkYe0IJWv2dJtYS5leierPjvdon6qI5le/EWg9gSimJX/VrxM9BlHCH3S1
BPONX2nsD5PHVH1Ad8toPI4orz0I1psRj5EFDL5UeIqjiWwwHK/Z32jXFZMLDIrB
ACPvkaBYW4gBywogPTGL5C7uamwQc+RJxWWVC7eGDqQlZB9HjwTOxH1zq2nHlHRg
HFYoVW7QedsxgvgqWeuvLyhLEd3wsXXMAQIDAQAB
-----END RSA PUBLIC KEY-----`

type publicKey struct {
	N *big.Int
	E int
}

func main() {
	var pub publicKey
	_, err := asn1.UnmarshalWithParams([]byte(der), &pub, "")
	if err != nil {
		fmt.Println(err)
		return
	}
	fmt.Println(pub.N)
}
$ tinygo build -o main.wasm -target=wasi main.go 
$ wazero run main.wasm
asn1: structure error: tags don't match (16 vs {class:0 tag:13 length:45 isCompound:true}) {optional:false explicit:false application:false private:false defaultValue:<nil> tag:<nil> stringType:0 timeType:0 set:false omitEmpty:false} publicKey @2
$ go run main.go
asn1: structure error: tags don't match (16 vs {class:0 tag:13 length:45 isCompound:true}) {optional:false explicit:false application:false private:false defaultValue:<nil> tag:<nil> stringType:0 timeType:0 set:false omitEmpty:false} publicKey @2
```

deadprogram, 2023-06-14:
Yes, thanks everyone!

closed 2023-06-14 by deadprogram