tinygo-org/tinygo #2451: Interrupt Callback Pin Parameter

fiurthorn, 2021-12-31:
Device: raspberry pi pico
Version: tinygo version 0.21.0 linux/amd64 (using go version go1.17.5 and LLVM version 11.0.0)

i expected that i con read the button state via the pin parameter

```go
var (
  led    = machine.GP2
  button = machine.GP3
)

button.Configure(machine.PinConfig{Mode: machine.PinInputPullup})
led.Configure(machine.PinConfig{Mode: machine.PinOutput})

button.SetInterrupt(machine.PinFalling|machine.PinRising, func(p machine.Pin) {
  // work
  led.Set(button.Get())
  // don't work
  led.Set(p.Get())
})
```

aykevl, 2022-01-19:
Yes, that's definitely a bug.

charathram, 2022-06-25:
Definitely seems to be a bug on the Pico. Is there a plan to fix this?

kenbell, 2022-06-25:
This is the current code to handle pin interrupts for rp2040 - looks like some TODO stuff to fix up...
```go
// gpioHandleInterrupt finds the corresponding pin for the interrupt.
// C SDK equivalent of gpio_irq_handler
func gpioHandleInterrupt(intr interrupt.Interrupt) {
	// panic("END") // if program is not ended here rp2040 will call interrupt again when finished, a vicious spin cycle.
	core := CurrentCore()
	callback := pinCallbacks[core]
	if callback != nil {
		// TODO fix gpio acquisition (see below)
		// For now all callbacks get pin 255 (nonexistent).
		callback(0xff)
	}
	var gpio Pin
	for gpio = 0; gpio < _NUMBANK0_GPIOS; gpio++ {
		// Acknowledge all GPIO interrupts for now
		// since we are yet unable to acquire interrupt status
		gpio.acknowledgeInterrupt(0xff) // TODO fix status get. For now we acknowledge all pending interrupts.
		// Commented code below from C SDK not working.
		// statreg := base.intS[gpio>>3]
		// change := getIntChange(gpio, statreg.Get())
		// if change != 0 {
		// 	gpio.acknowledgeInterrupt(change)
		// 	if callback != nil {
		// 		callback(gpio)
		// 		return
		// 	} else {
		// 		panic("unset callback in handler")
		// 	}
		// }
	}
}
```

kenbell, 2022-06-25:
This PR should fix things up: https://github.com/tinygo-org/tinygo/pull/2931

deadprogram, 2022-07-09:
Closing since this has been released. Thanks!

closed 2022-07-09 by deadprogram