tinygo-org/tinygo #2452: SetInterrupt Callback with PinChange

fiurthorn, 2021-12-31:
I would find it interesting to have the event trigger passed along with it.

```go
var (
	led    = machine.GP2
	button = machine.GP3
)

button.Configure(machine.PinConfig{Mode: machine.PinInputPullup})
led.Configure(machine.PinConfig{Mode: machine.PinOutput})

var start time.Time 
button.SetInterrupt(machine.PinFalling|machine.PinRising, func(p machine.Pin, c machine.PinChange) {
	if c == machine.PinFalling {
		start = time.Now()
	} else if c == machine.PinRising {
		d := time.Since(start).Seconds()
		if d > 5 {
			led.High()
		}
	}
})
```

bmentink, 2022-02-26:
I vote for this inhancement as well 👍

aykevl, 2022-03-01:
It does seem reasonable, but there are two issues:

* This would be a breaking change. This may or may not be acceptable, but it is not something to be done lightly.
* I don't think all chips support this feature. I quickly looked at the Atmel/Microchip SAM D21 series (popular in "M0" branded boards) and they don't seem to differentiate between the various interrupt sources.

So, not sure about this.

> My use case is a sensorless BLDC motor control

Oof, that's a rather heavy duty use case! It may be reasonable to do this by directly manipulating registers, just so you know exactly what's happening. In fact, I believe some chips have direct support for these motors but it's certainly out of my area of expertise so can't really help you with this :)

bmentink, 2022-03-01:
@aykevl 
Not that heavy duty use case, if it can be implemented on an 8-bit Arduino Uno  in C ... :)

fiurthorn, 2023-08-30:
it the normal use case of an JK latch (Flipflop).