tinygo-org/tinygo #2456: Invalid bitcast LLVM ERROR: Broken function

BoskyWSMFN, 2021-12-31:
Hello!
Found an issue using coroutines with avr:
running following command
``tinygo build -gc=conservative -scheduler=coroutines -target=arduino -o ./main.hex ./main.go``
with main.go containing
```
package main

import "time"

func yetAnotherAsyncFunc() {
	for {
		time.Sleep(time.Second)
	}
}

func main() {
	go yetAnotherAsyncFunc()
	println("Hello!")
}
```
leads to this error:
```
Invalid bitcast
i8* bitcast (void (i8*, i8*) addrspace(1)* @main.yetAnotherAsyncFunc to i8*)
LLVM ERROR: Broken function
SIGABRT: abort
PC=0x7fa581a50fbb m=5 sigcode=18446744073709551610
```
However I managed to build this code with commenting the `import` section.

Thank you and happy holidays!

dgryski, 2021-12-31:
Note that coroutines are buggy and are being removed:
https://github.com/tinygo-org/tinygo/issues/2174
https://github.com/tinygo-org/tinygo/pull/2448

BoskyWSMFN, 2021-12-31:
Thank you for the information! Didn't read all closed issues yet.
Since coroutines are being removed is there a "tiny-go" way to use go routines with AVR MCUs? I couldn't manage the tasks scheduler work unfortunately.

niaow, 2021-12-31:
The `tasks` scheduler is supposed to work. What issue did you run into with it?

niaow, 2021-12-31:
One thing to note with that program you provided is that Go considers the program "done" as soon as `main` exits.

BoskyWSMFN, 2021-12-31:
> The `tasks` scheduler is supposed to work. What issue did you run into with it?

Blinking user LED on Mega and Nano boards without any other signs of getting to `main`. For example - the code I provided above.

niaow, 2021-12-31:
> Blinking user LED on Mega and Nano boards without any other signs of getting to main. For example - the code I provided above.

Hmm. . . I don't have a Mega/Nano to test with, but it seems to have worked on my Uno - it prints hello and stops correctly.
Can you provide your blinking code (the provided code does not blink).

BoskyWSMFN, 2021-12-31:
> Hmm. . . I don't have a Mega/Nano to test with, but it seems to have worked on my Uno - it prints hello and stops correctly.
> Can you provide your blinking code (the provided code does not blink).

Well... The code above does blink on my mega. Tested just now. Looks like some kind of error code but I can't find anything.
May be there is an issue with a bootloader and fuses...

niaow, 2021-12-31:
Oh, I see. I guess I just misunderstood what you were saying.

closed 2021-12-31 by BoskyWSMFN

niaow, 2021-12-31:
Did you find the problem?

BoskyWSMFN, 2022-01-01:
> Did you find the problem?

Not yet. I'll try to later and reopen this issue when there is something to show.