tinygo-org/tinygo #2463: "GOOS=windows tinygo test" fails while invoking Wine installed with winehq installer on Mac OS 11.6?

dkegel-fastly, 2022-01-01:
tinygo's wine support is great, but it has a little trouble with the wine installed from the packages at https://wiki.winehq.org/MacOS, but manually running wine64 gets around the problem for me.

Maybe the package still includes some 32 bit stuff which no longer works on current MacOS.

Here's how to reproduce (the first two lines are from memory and might be off a bit):

```
$ brew tap homebrew/cask-versions
$ brew install --cask --no-quarantine wine-stable
$ GOOS=windows tinygo test -v os
wine: could not load ntdll.so: dlopen(/usr/local/bin/../lib64/wine/ntdll.so, 2): image not found
FAIL	os	0.226s
FAIL
$ GOOS=windows tinygo test -c os
$ wine64 os.test -test.v
=== RUN   TestConsistentEnviron
...
```

aykevl, 2022-01-19:
So basically you just need to call `wine64` instead of `wine`? That would be easy to fix. You can replace the following line:

https://github.com/tinygo-org/tinygo/blob/38cd5b80ec8d84475895a4cc2cafb3e52f053889/compileopts/target.go#L329

With

```go
if runtime.GOOS == "darwin"
	spec.Emulator = []string{"wine64"}
} else {
	spec.Emulator = []string{"wine"}
}
```

Feel free to send a PR if that works better on MacOS.

dkegel-fastly, 2022-01-19:
I suspect it'll want to try both (to handle different sorts of wine installs), so some finesse may be required.  I'll have a look sometime.