tinygo-org/tinygo #2464: runtime (avr): timer code fails with -opt=1 or during init()

niaow, 2022-01-01:
When debugging `testdata/goroutine.go`, I came up with this minimal test case (the loop isn't necessary, but it makes things easier to test on the actual hardware):
```
package main

import "time"

func init() {
        time.Sleep(time.Millisecond)
}

func main() {
        for {
                println("hello")
        }
}
```

If you check the scheduler output, the chip goes to sleep at which point simavr recognizes that it cannot wake up. This also reproduced on real hardware (my Arduino Uno). It reproduces with or without a scheduler.

Another way to replicate this is to run `examples/blinky1` with `-opt=1`.

aykevl, 2022-01-01:
I think I know the cause, will make a PR now.

aykevl, 2022-01-01:
Fix: https://github.com/tinygo-org/tinygo/pull/2470

PR #2470 (2022-01-02): avr: fix time.Sleep() in init code
In the early days of TinyGo, the idea of `postinit` was to enable interrupts only after initializers have run. Which kind of makes sense... except that `time.Sleep` is allowed in init code and `time.Sleep` requires interrupts to be enabled. Therefore, interrupts must be enabled while initializers are being run.

This commit simply moves the enabling of interrupts to a point right before running package initializers. It also removes `runtime.postinit`, which is not necessary anymore (and was only used on AVR).

This fixes #2464.
closed 2022-01-26 by deadprogram