tinygo-org/tinygo #2466: assert-test-linux "test stdlib packages" reports success despite test failing to link?

dkegel-fastly, 2022-01-01:
In the CI results for https://github.com/tinygo-org/tinygo/pull/2462, 
in the assert-test-linux CI run, 
in the "test stdlib packages" step, there is an error linking one test, but the build does not terminate.

Log fragment:
```
tinygo:ld.lld: error: undefined symbol: syscall.RawSyscall
>>> referenced by zsyscall_linux_amd64.go:1661 (/opt/hostedtoolcache/go/1.17.5/x64/src/syscall/zsyscall_linux_amd64.go:1661)
>>>               /tmp/tinygo2693500925/main.o:(os_test.TestSmokePipe)
failed to run tool: ld.lld
```

This bug isn't about the link failure; it's about the test suite passing despite the link failure.

niaow, 2022-01-02:
I think I see what the problem is here.

dkegel-fastly, 2022-01-02:
Thanks!  Filed https://github.com/tinygo-org/tinygo/issues/2474 for the missing syscall magic.

closed 2022-01-26 by deadprogram