tinygo-org/tinygo #2467: time.After(): undefined symbol: time.startTimer

fiurthorn, 2022-01-01:
Use of time.After() leads to an undefined symbol:

```go
func timer(start, stop chan struct{}) {
	var span time.Time
	select {
	case <-start:
		span = time.Now()
	case <-stop:
		span = time.Time{}
	case <-time.After(250 * time.Microsecond):
		if span.Second() > 5 {
			led.High()
		}
	}
}
```

```bash
fiurthorn@midgard:~/workspace/tiny$ tinygo flash -target=pico ./main.go
tinygo:ld.lld: error: undefined symbol: time.startTimer
>>> referenced by sleep.go:96 (/opt/go/src/time/sleep.go:96)
>>>               /tmp/tinygo3840351086/main.o:(main.timer$gowrapper)
failed to run tool: ld.lld
error: failed to link /tmp/tinygo3840351086/main: exit status 1
```

niaow, 2022-01-02:
This is a duplicate of #1037

aykevl, 2022-01-19:
Yes, going to close this in favor of #1037. Please follow that issue instead.

closed 2022-01-19 by aykevl