tinygo-org/tinygo #2474: linux: undefined symbol: syscall.RawSyscall

dkegel-fastly, 2022-01-02:
In the CI results for #2462,
in the assert-test-linux CI run,
in the "test stdlib packages" step, there is an error linking one test:

Log fragment:

```
tinygo:ld.lld: error: undefined symbol: syscall.RawSyscall
>>> referenced by zsyscall_linux_amd64.go:1661 (/opt/hostedtoolcache/go/1.17.5/x64/src/syscall/zsyscall_linux_amd64.go:1661)
>>>               /tmp/tinygo2693500925/main.o:(os_test.TestSmokePipe)
failed to run tool: ld.lld
```

This also reproduces locally on linux.

Upstream, this is from assembly, e.g.:
./src/cmd/vendor/golang.org/x/sys/unix/asm_linux_arm64.s:TEXT ·RawSyscall(SB),NOSPLIT,$0-56

In tinygo, this kind of thing appears to be handled by compiler/syscall.go.

aykevl, 2022-01-19:
Yes, I _think_ it would be safe to implement `syscall.RawSyscall*` functions as aliases of `syscall.Syscall*` functions (see https://stackoverflow.com/questions/16977988/details-of-syscall-rawsyscall-syscall-syscall-in-go).
This might work:

```diff
diff --git a/compiler/compiler.go b/compiler/compiler.go
index 86b20306..f0a634c1 100644
--- a/compiler/compiler.go
+++ b/compiler/compiler.go
@@ -1438,7 +1438,7 @@ func (b *builder) createFunctionCall(instr *ssa.CallCommon) (llvm.Value, error)
                        return b.emitSV64Call(instr.Args)
                case strings.HasPrefix(name, "(device/riscv.CSR)."):
                        return b.emitCSROperation(instr)
-               case strings.HasPrefix(name, "syscall.Syscall"):
+               case strings.HasPrefix(name, "syscall.Syscall") || strings.HasPrefix(name, "syscall.RawSyscall"):
                        return b.createSyscall(instr)
                case strings.HasPrefix(name, "syscall.rawSyscallNoError"):
                        return b.createRawSyscallNoError(instr)
```

If that fixes the issue, feel free to send a PR.

dkegel-fastly, 2022-01-20:
Seems to work.  Appended it as a commit to https://github.com/tinygo-org/tinygo/pull/2462, which turns it a nice pretty green.