tinygo-org/tinygo #2477: undefined symbol: __aeabi_unwind_cpp_pr0 for ARM target

QuLogic, 2022-01-02:
With LLVM 12 support landed, I'm trying to update the Fedora packages to be usable. I'm running in to some trouble running the arm smoke test:
```
$ make smoketest STM32=0 XTENSA=0
...
GOOS=linux GOARCH=arm /builddir/build/BUILDROOT/tinygo-0.21.0-2.fc34.aarch64/usr/bin/tinygo build -size short -o test.elf       ./testdata/cgo
ld.lld: error: undefined symbol: __aeabi_unwind_cpp_pr0
>>> referenced by divsi3.c
>>>               divsi3.c.o:(.ARM.exidx.text.__divsi3+0x0) in archive /builddir/build/BUILD/tinygo-0.21.0/tinygo.ISH612/tinygo/compiler-rt-armv6-unknown-linux-gnueabihf-generic/lib.a
>>> referenced by udivsi3.c
>>>               udivsi3.c.o:(.ARM.exidx.text.__udivsi3+0x0) in archive /builddir/build/BUILD/tinygo-0.21.0/tinygo.ISH612/tinygo/compiler-rt-armv6-unknown-linux-gnueabihf-generic/lib.a
>>> referenced by udivmodsi4.c
>>>               udivmodsi4.c.o:(.ARM.exidx.text.__udivmodsi4+0x0) in archive /builddir/build/BUILD/tinygo-0.21.0/tinygo.ISH612/tinygo/compiler-rt-armv6-unknown-linux-gnueabihf-generic/lib.a
>>> referenced 64 more times
error: failed to link /tmp/tinygo570417244/main: exit status 1
```

Formerly, I only ran native Linux tests (i.e., `go test` ones) on the matching platform (e.g., amd64 on x86_64), because Fedora did not have cross-compiled libc available even though there was the cross-compiler. Now that tinygo uses its own copy of musl, running tests for all platforms might be feasible again. However, the smoke test shows that arm is still not quite working. I see that #2155 is supposed to fix something similar, but it's in 0.21.0 already.

aykevl, 2022-01-03:
Hmm, I'm assuming you've built with the system LLVM in Fedora?
I suspect this build has a few defaults that are different from upstream LLVM. Especially something related to stack unwinding (exceptions, etc). I can't reproduce it locally, using the Debian variant of LLVM:

```
GOOS=linux GOARCH=arm tinygo build -size short -o test.elf       ./testdata/cgo/
   code    data     bss |   flash     ram
  15668     308    1424 |   15976    1732
```

(built with `go install -tags=llvm12`)

aykevl, 2022-01-03:
> Formerly, I only ran native Linux tests (i.e., `go test` ones) on the matching platform (e.g., amd64 on x86_64), because Fedora did not have cross-compiled libc available even though there was the [...]

Yes, TinyGo is now always a cross compiler. It is very likely that the exact same issue will occur on a native ARM version of Fedora.

QuLogic, 2022-01-03:
> Hmm, I'm assuming you've built with the system LLVM in Fedora?
> I suspect this build has a few defaults that are different from upstream LLVM. Especially something related to stack unwinding (exceptions, etc).

Yes, it is as far as I know a vanilla LLVM. The [only patch](https://src.fedoraproject.org/rpms/llvm/blob/f34/f/llvm.spec#_65) appears to be fixing a test. There is `-fexceptions` in the `CFLAGS` at LLVM build time; I'm not sure if that translates to it being on by default at runtime, too.

QuLogic, 2022-01-03:
Ah, on the other hand, it looks like `clang` has more patches, and [this one enable `-funwind-tables` everywhere](https://src.fedoraproject.org/rpms/clang/blob/f34/f/0002-PATCH-clang-Make-funwind-tables-the-default-on-all-a.patch) I tried to add `-funwind-tables` here:
```
diff --git a/builder/library.go b/builder/library.go
index 73079785..012e4053 100644
--- a/builder/library.go
+++ b/builder/library.go
@@ -120,7 +120,7 @@ func (l *Library) load(config *compileopts.Config, tmpdir string) (job *compileJ
                }
        }
        if strings.HasPrefix(target, "arm") || strings.HasPrefix(target, "thumb") {
-               args = append(args, "-fshort-enums", "-fomit-frame-pointer", "-mfloat-abi=soft")
+               args = append(args, "-fshort-enums", "-fomit-frame-pointer", "-mfloat-abi=soft", "-fno-unwind-tables")
        }
        if strings.HasPrefix(target, "riscv32-") {
                args = append(args, "-march=rv32imac", "-mabi=ilp32", "-fforce-enable-int128")
```
but maybe that's the wrong place?

QuLogic, 2022-01-03:
> Yes, TinyGo is now always a cross compiler. It is very likely that the exact same issue will occur on a native ARM version of Fedora.

That's good; I've removed my test-only-on-native patch, and I can confirm tests of 64-bit/32-bit x86 + aarch64 all work on x84_64 properly. There may be some other issues doing the opposite that need to be worked out, though it may be qemu that's bugged.

QuLogic, 2022-01-03:
> I tried to add `-fno-unwind-tables` here:
> ...
> but maybe that's the wrong place?

Oh, I should have read the output a bit closer; it's now complaining about `main.c` instead of the libc files, i.e., that did work, it just didn't hit everything that went through `clang`.

aykevl, 2022-01-03:
> Ah, on the other hand, it looks like `clang` has more patches, and [this one enables `-funwind-tables` everywhere](https://src.fedoraproject.org/rpms/clang/blob/f34/f/0002-PATCH-clang-Make-funwind-tables-the-default-on-all-a.patch).

I was expecting something like this.

> but maybe that's the wrong place?

That is the correct place. But it's only one of two places, as you've found out :)

The other place that needs this flag is here (`spec.CFlags`):

https://github.com/tinygo-org/tinygo/blob/06df31944c6af706d6cb8b30a2e92d19060436d2/compileopts/target.go#L257-L266

aykevl, 2022-02-05:
This was fixed in the latest release.

closed 2022-02-05 by aykevl