tinygo-org/tinygo #2480: os.IsNotExist does not work properly on Windows?

dkegel-fastly, 2022-01-03:
dev's TestStatBadDir in os_anyos_test.go currenly skips on Windows because os.IsNotExist doesn't seem to work there.

It appears that syscall_windows.go's Errno.Is() function appears to be broken for us; the following kludge lets the test pass:

```
diff a/src/os/errors.go b/src/os/errors.go
@@ -60,12 +61,34 @@ func underlyingErrorIs(err, target error) bool {
        }
        // To preserve prior behavior, only examine syscall errors.
        e, ok := err.(syscall.Errno)
-       return ok && e.Is(target)
+       if !ok {
+               return false
+       }
+       if runtime.GOOS == "windows" {
+               // syscall_windows.go's Errno.Is() function appears to be broken for us, so recapitulate it here, badly and partially.
+               switch target {
+                       //case ErrPermission:
+                       //      return e == ERROR_ACCESS_DENIED
+                       //case ErrExist:
+                       //      return e == ERROR_ALREADY_EXISTS ||
+                       //              e == ERROR_DIR_NOT_EMPTY ||
+                       //              e == ERROR_FILE_EXISTS
+                       case ErrNotExist:
+                               return e.Error() == "File not found." ||
+                                       e.Error() == "Path not found."
+                       //      return e == ERROR_FILE_NOT_FOUND ||
+                       //              e == _ERROR_BAD_NETPATH ||
+                       //              e == ERROR_PATH_NOT_FOUND
+               }
+       }
+       return e.Is(target)
 }
```

dkegel-fastly, 2022-04-28:
Nope.  Still present, verified on dev that the test fails if I remove the exception for windows.

deadprogram, 2022-04-28:
OK thanks for checking.

closed 2026-11-05 by deadprogram