tinygo-org/tinygo #2483: TestClearenv flaky?

dkegel-fastly, 2022-01-04:
I've seen this error twice now:

```
--- FAIL: TestClearenv
    Setenv("", "\x00\x00\x00\x00\x00\x00P\x00\x00\x00\x00\x00") failed during reset: setenv: errno 22
```

This time it was on my mac, can't remember where it was last time.

dkegel-fastly, 2022-01-22:
This is now my most frequent flaky test.

dgryski, 2022-01-24:
This is bizarre that this test is flaky.  Almost points to some bigger underlying issue.  I'll maybe take a look.

dkegel-fastly, 2022-01-24:
The test expects the return value of Environ() to remain valid (i.e. noncorrupt) after the environment is modified, as on upstream go, but it is disappointed because Tinygo does not copy the environment.

env.go needs a bit of work.

dankegel, 2022-03-12:
Fixed by https://github.com/tinygo-org/tinygo/pull/2647

closed 2022-04-29 by deadprogram