tinygo-org/tinygo #2486: make fails on macos m1 arm

exyzzy, 2022-01-05:
Per build instructions:

git clone --recursive https://github.com/tinygo-org/tinygo.git
cd tinygo
make llvm-build         # all good up to here
make                           # this breaks

couple thousand lines of errors, but the summary seems to be:
 ...building for macOS-x86_64 but attempting to link with file built for macOS-arm64
...Undefined symbols for architecture x86_64:
...ld: symbol(s) not found for architecture x86_64

aykevl, 2022-01-19:
I think most of us don't have the hardware to reproduce this issue. So can you give a bit more details? In particular, can you show the full error message, or if it is long at least part of the raw error message?

I suspect one of the build tools you're using is actually a x86_64 binary. Have you checked that the `go` binary is really an arm64 binary?

dkegel-fastly, 2022-07-13:
make no longer fails.

closed 2022-07-13 by dkegel-fastly