tinygo-org/tinygo #2492: "LLVM ERROR: Cannot select: t3: i16 = JumpTable" in CI on AVR target

dkegel-fastly, 2022-01-06:
Got an ugly llvm crash in CI just now: https://github.com/tinygo-org/tinygo/runs/4722070252

```
=== CONT  TestBuild/AVR/goroutines.go
LLVM ERROR: Cannot select: t3: i16 = JumpTable<0>
In function: machine.AdjustMonotonicTimer
SIGABRT: abort
...
FAIL	github.com/tinygo-org/tinygo	62.689s
make: *** [Makefile:196: test] Error 1
```

Happened again in https://github.com/tinygo-org/tinygo/runs/4722481707
```
=== CONT  TestBuild/AVR/slice.go
LLVM ERROR: Cannot select: t3: i16 = JumpTable<0>
In function: runtime.alloc
SIGABRT: abort
```

Seems to happen at about 2m10s into Test TinyGo; successful runs seem to take longer to get to the same point.

Here’s a good run (a retry of the previous one): https://github.com/tinygo-org/tinygo/runs/4722602927

No idea why it doesn't happen every time.  Different software or hardware on different github ci nodes?

dkegel-fastly, 2022-01-06:
AVR may have some dirty laundry: https://github.com/llvm/llvm-project/issues/31494#issuecomment-980975376

dkegel-fastly, 2022-01-06:
Happened again in https://github.com/tinygo-org/tinygo/runs/4729218984, again at 2 minutes 16 seconds, line 413.
```
=== CONT  TestBuild/AVR/slice.go
LLVM ERROR: Cannot select: t3: i16 = JumpTable<0>
In function: runtime.alloc
SIGABRT: abort
```

dkegel-fastly, 2022-01-06:
FWIW, I snuck a "cat /proc/cpuinfo" into the top of the apt-get install step in https://github.com/tinygo-org/tinygo/pull/2487, the pull request that fails half the time with this problem.

On the first run, it said the cpu was a 2 core Xeon 8370C (so a small slice of a 32 core chip), and it did not hit this bug.

aykevl, 2022-01-06:
> AVR may have some dirty laundry: [llvm/llvm-project#31494 (comment)](https://github.com/llvm/llvm-project/issues/31494#issuecomment-980975376)

Not anymore :) see https://reviews.llvm.org/D107853#3224989

My guess would be a concurrency issue, so happening only sometimes. If someone can reproduce this locally that would be a very big help towards fixing this issue (especially if I manage to reproduce this on my own laptop).
What is especially strange is that `JumpTable` should not be used anyway, because it is disabled. However, I have a slight suspicion this commit might have changed things: https://github.com/llvm/llvm-project/commit/567f6c150dd726c2ec9175ba7e7a6a954e69d0f6 - although I don't really know how.

dkegel-fastly, 2022-01-09:
I seem to be able to reproduce something similar locally on my mac with the command

```go test  -run TestTest```

on https://github.com/tinygo-org/tinygo/pull/2500, on a hex core macbook pro,  using go version go1.17.6 and LLVM version 12.0.1

Log:

```
...
=== PAUSE TestTest/EmulatedRISCV/Fail
=== CONT  TestTest/EmulatedRISCV/Pass
=== CONT  TestTest/EmulatedRISCV/Fail
=== CONT  TestTest/Host/Nothing
    main_test.go:612: ?         github.com/tinygo-org/tinygo/tests/testing/nothing      [no test files]
LLVM ERROR: Cannot select: 0x614eba0: i32 = JumpTable<0>
In function: (reflect.Value).Len
SIGABRT: abort
PC=0x7fff2075d92e m=10 sigcode=0
...

dkegel-fastly, 2022-01-09:
I'm preserved the failing commit as the tip of https://github.com/dkegel-fastly/tinygo/tree/dkegel-llvm12-error-2492

It seems to fail only locally, not in CI.

aykevl, 2022-01-21:
I wonder if we could perhaps try to build LLVM with AddressSanitizer or ThreadSanitizer enabled? Not sure whether that works together with Go though. But those sanitizers might be able to find something.

aykevl, 2022-01-21:
Or if this becomes reproducible on Linux, that would be awesome because then I can investigate this myself :)

deadprogram, 2024-12-04:
@aykevl @dkegel-fastly is this issue still valid?

deadprogram, 2024-12-04:
OK, in that case closing. Thank you!

closed 2024-12-04 by deadprogram